genodelabs/genode #1334: Disable certificate checking when downloading ports

chelmuth, 2015-01-02:
Currently, we have an issue with red.libssh.org and wget/curl that can't validate the certificate information of the site. Firefox has no such problem. After some more investigations I discovered the problem from an _intermediate certificate_ from StartCom used to sign the server/node certificate for red.libssh.org (also discussed [here](http://security.stackexchange.com/questions/57535/why-do-i-need-to-install-the-startcom-intermediate-certificate)).

The issue could be fixed on server and client side by installing the intermediate certificate. But, I'd rather like to question if we should use certificate checking (or even HTTPS) when downloading third-party code as we check the archive integrity with our checksums anyway. Opinions?

cnuke, 2015-01-02:
:+1: for using HTTP only for now.

In the long term, more server will probably switch to using HTTPS only (hence for obvious but arguably good reasons) I guess — so the issue with certificate checking will not go away.

nfeske, 2015-01-03:
For protecting the integrity of the downloaded data, HTTPS does not add much value. The checksums are sufficient for that. The only advantage that HTTPS has over HTTP is that the mere information about which files are downloaded from the server is protected. I don't think that we should jump through hoops to hide this bit of information.

I agree to stay with HTTP for downloading libssh.

chelmuth, 2015-01-06:
Unfortunately, the case of libssh is just like @cnuke apprehended: The download is available via HTTPS only. Even after the server admin fixed the certificate issue after I approached him, we still miss the SNI support in wget < 3.14 and, therefore, need to add `--no-check-certificate` for a successful download on Ubuntu 12.04 LTS :cry:

commit 11a513ac (2015-01-06): prepare_port: disable certificate checking in wget

Some downloads are available via HTTPS only, but wget < 3.14 does not
support server-name identification, which is used by some sites. So, we
disable certificate checking in wget and check the validity of the
download via SIG or SHA.

Fixes #1334.
closed 2015-01-08 by chelmuth