genodelabs/genode #1338: nova: remove timer delay heuristic

alex-ab, 2015-01-06:


nfeske, 2015-01-06:
@alex-ab Hats off for finding the bug!

What do you think about documenting the symptom and the actual problem here for future reference?

Let me just annotate this issue with the link to the actual fix: https://github.com/alex-ab/NOVA/commit/66350b61f17b475f05570505c4a1b23106fcb40a

alex-ab, 2015-01-06:
Sympton:

On some x86 platforms using base-nova, it happens that the system seems to slow down dramatically for some unclear reasons. As soon as a VM (either Windows 7 and - reported by @chrism333 - also Android x86) boots up with VirtualBox as VMM, the Genode host becomes in-responsive, e.g. when using Nitpicker the mouse starts "jumping". Also moving the VM and VMM to another CPU to reduce load on the first CPU does not help (as reported by @chrism333).

Debugging the problem:

The slow down was observed using the commits from #813 which effictively accumulate all CPU utilization times of mostly all threads in the Genode setup. From the output of the tool one could observe that the most time is spent in the threads of services (core/init/drivers/nitpicker) with the highest priorities. From the observation it was clear that the system is still somehow working, but some applications as the Guest VM effectively are never scheduled anymore. 

When enabling some more VM exits we could isolate the effect to some rare VM exits (Windows as guest VM) like wbinvd. Further debugging output printed by the nova kernel showed that the *host* cr0 register shortly before the wirst wbinvd VM exit changed. The affected bit was the 30. - the cache disable (CD) bit. So - somehow the guest managed to switch of the caches of the host system. After further debugging and conversation with @udosteinberg we finally found the bug. According to the Intel Spec - chapter 27.5.1 Loading Host Control Registers, Debug Registers, MSRs - not all bits of all control registers, e.g. CD bit of cr0, gets reloaded by the hardware automatically. Effectively that means the CD bit was taken from the guest and used by the host, afaik.

From this point in time everthing was somehow "slow". The effect vanished as soon as the guest VM switched caches on (disabling the CD bit). Unfortunately this may happen never when a Genode setup was used where the VMM and the VM runs on the lowest priority.

nfeske, 2015-01-06:
@alex-ab Thank you for this great summary!

commit 06d3bb84 (2015-01-06): nova: remove timer delay heuristic

The commit uses a fixed kernel branch (r8), which fixes a caching bug
observable in the Genode host. The quirk detecting the circumstance in the
timer service is obsolete now and is removed.

Fixes #1338
commit 8842ba2e (2015-01-06): nova: remove timer delay heuristic

The commit uses a fixed kernel branch (r8), which fixes a caching bug
observable in the Genode host. The quirk detecting the circumstance in the
timer service is obsolete now and is removed.

Fixes #1338
closed 2015-01-08 by alex-ab