genodelabs/genode #1340: hw vea9x4: let the timer SP804 0/1 be secure

m-stein, 2015-01-06:


chelmuth, 2015-01-07:
My hope was your commit would fix our nightly issues, but all hw_vea9x4_tz tests still fail with something like
```
[2015-01-07 05:54:58] kernel initialized
[2015-01-07 05:54:58] Genode 14.11
[2015-01-07 05:54:58] int main(): --- create local services ---
[2015-01-07 05:54:58] int main(): --- start init ---
[2015-01-07 05:54:58] int main(): transferred 29 MB to init
[2015-01-07 05:54:58] int main(): --- init created, waiting for exit condition ---
[2015-01-07 05:54:58] no RM attachment (faulter 19090 with IP 4801edc0 attempts to read from address 480fefb4)
[2015-01-07 05:54:58] void Genode::Pager_object::unresolved_page_fault_occurred(): not implemented
[2015-01-07 05:54:58] unknown signal context
```
Did you address hw_vea9x4_tz or hw_vea9x4 only? Any successful run of hw_vea9x4_tz on your side?

chelmuth, 2015-01-07:
Hm, the commit in #1337 was not part of staging this night. Does it address the above issue? I just merged it into staging.

m-stein, 2015-01-07:
@chelmuth: Jep, that should've been the problem.

commit ba321c20 (2015-01-06): hw vea9x4: let userland timer SP804 0/1 be secure

Declaring the SP804 0/1 module and its interrupt to be non-secure prevents the
secure Genode from receiving the interrupt and hence the timer driver in the
secure Genode doesn't work.

Fixes #1340
closed 2015-01-12 by m-stein