genodelabs/genode #1342: libports: add gtest (the Google C++ testing framework)

rite, 2015-01-06:
> Google's framework for writing C++ tests on a variety of platforms (Linux, Mac OS X, Windows, Cygwin, Windows CE, and Symbian). Based on the xUnit architecture. Supports automatic test discovery, a rich set of assertions, user-defined assertions, death tests, fatal and non-fatal failures, value- and type-parameterized tests, various options for running the tests, and XML test report generation.

https://code.google.com/p/googletest/

To get some first hands-on experience with genode, I tried to port gtest. So far it was straight forward and most of the tests succeed. I've included two run scripts. gtest.run executes the tests of the testing framework itself and gtest-samples.run executes a simple demo test application.

There are some open points:

* All three patches are workarounds. Any suggestions for proper solutions?

* Some tests are filtered because they break the execution of subsequent tests (see gtest_filter argument in gtest.run).

* From the remaining 699 tests, 3 tests fail.

* Because rtc_drv is required, the run scripts work on base-nova but not on base-linux.

chelmuth, 2015-01-06:
The dev branch is http://github.com/rite/genode/commits/gtest.

chelmuth, 2015-01-06:
Once #1344 is closed, the run script should work on base-linux again.

rite, 2015-01-07:
@chelmuth Thanks!
I forgot to mention that the gtest-filepath.patch used in the port file might get obsolete when #1330 is closed.

chelmuth, 2017-01-17:
I now have three versions of this developments applying to [16.11](https://github.com/chelmuth/genode/commits/googletest-16.11), current [master](https://github.com/chelmuth/genode/commits/googletest-master), and current [staging](https://github.com/chelmuth/genode/commits/googletest-staging). Also, I propose to merge the branch when next appropriate. Opinions?

nfeske, 2017-01-18:
@chelmuth Thanks for reviving the issue. This port would fit much better in the genode-world repository though. I ultimately want to reduce the number of 3rd party ports from the main genode repository. This applies to all ports that are not closely coupled to Genode, e.g., dosbox, lighttpd, fuse, sdl, and even qt5. The ports within the regular genode repository should be restricted to 3rd-party software that needs substantial customization or continuous adaptation for Genode, i.e., device drivers, virtualbox, libc, or the various kernels.

Hosting 3rd-party software in genode-world has a number of advantages. Foremost, it substantially lowers the barrier of entry for maintainers outside of Genode Labs (e.g., there is no contributor's agreement to sign) to participate. Second, it cultivates the clean separation of the base framework from application-level components that are less critical. E.g., by merging commits to genode-world, the genode base framework cannot be negatively affected.

Note that by moving existing ports from the genode repository to genode-world, I do not want to imply to loosen the quality assurance of such ports. In contrary, I would like to successively extend our current build and integration tests (that we routinely apply to the genode repository) to the genode-world repository.

rite, 2017-01-18:
I see the reasons behind moving 3rd party ports to the genode-world repository. And gtest undoubtedly qualifies as a "component that does not fall in the narrow scope of the Genode OS Framework".

My only concerns are related to the configuration management that needs to be addressed when having two (more or less tightly) related repositories. Especially when you plan to provide long-term support for certain Genode OS Framework releases.

That said, I don't see that this would be an immediate problem for the gtest library right now. So I don't mind whether the the gtest library is provided by the genode or the genode-world repository.

nfeske, 2017-01-18:
@rite Thanks for sharing your perspective on the topic.

nfeske, 2017-06-29:
Merged to genode-world. Thanks @senier!

closed 2017-06-29 by nfeske