genodelabs/genode #1343: nova: starting/running 2+ VMs concurrently not possible

alex-ab, 2015-01-06:
It is kind of follow up issue of #1307. I hate to cite myself, but:

Seoul VMM:
```
I can confirm that the mouse input sometimes does not work. As soon as I start a second seoul instance the mouse input of the first instance seems to not processed anymore, but in the second instance the mouse works. When I start a third instance, the mouse of the second instance stops working ...
```

Additionally we have interferences when starting several Virtualbox VMMs in parallel. Both VMs (Windows guest) affect each other as soon as the second instance starts running. When the first VM ist booted completly up and the second starts, sometimes the first one produces a blue-screen, sometimes the second. When starting both VMs concurrently at the same time, sometimes one of it managed to boot up.
Sometimes both don't manage to boot up, one of it dies with some triple fault and immediatly after it the second one also dies, e.g.:

```
[ 0] Killed EC:0xffffffff82c9e280 SC:0xffffffff82e31660 V:0x3 CS:0x2b EIP:0x136c1e4 CR2:0x3d737 ERR:0x0 (PT not found)
[ 0] Killed EC:0xffffffff82e99280 SC:0xffffffff82e31660 V:0x30 CR0:0x80000011 CR3:0x0 CR4:0x0 (IPC Abort)
[ 0] Killed EC:0xffffffff82c9e700 SC:0xffffffff82c794a0 V:0x30 CR0:0x8001003b CR3:0x0 CR4:0x0 (IPC Timeout)
```
From the ouput you see one vcpu dying and immediatly after it the second vcpu dies - both vcpus are running in different protection domains and have no shared memory or devices assigned.

chelmuth, 2015-01-06:
I traced the issue back to commit e70c13ce1fc9391d4a086840a9d64674980a4f04 which enables the use of PCIDs in *base-nova/run/env* by removing `nopcid` fom the kernel command line. Re-adding the parameter enables the parallel execution of both Seoul and VirtualBox VMs. So, it seems we have issues with the PCID implementation in NOVA or miss required user-level handling in the context of virtual machines. @udosteinberg do you have suggestions to track this down?

nfeske, 2015-01-06:
@chelmuth Wow! You encircled the issue. That makes my day. :-)

alex-ab, 2015-01-06:
@chelmuth: :+1:  @nfeske: Let's go climbing ;-)

chelmuth, 2015-01-06:
Better not, but I'll get the well-deserved dinner soon ;-)

udosteinberg, 2015-01-07:
Can you send me a trace of all the PDs being created along with their PCIDs?

alex-ab, 2015-01-08:
In [https://github.com/alex-ab/genode/blob/f19634ec4843b054a68965c18c0f848eb2f07f81/trace-pd-did.txt trace-did.txt] you can see a trace of all created PDs and there did values (dumped in Pd constructor). In 
[https://github.com/alex-ab/genode/blob/f19634ec4843b054a68965c18c0f848eb2f07f81/trace-log.txt trace-log.txt] you may find the whole Genode trace, to better correlate user level actions with the kernel. So did 25 and did 30 are the PDs containing just a vcpu - so no other thread is running inside the PDs.

udosteinberg, 2015-01-08:
Do you have anything in your Genode-specific patches where you modify the user-accessible portion of an address space without going through ```Space_mem::update```?

The next step would be to instrument the part of ```Space_mem::update``` that does ```htlb.merge``` to mark the TLB dirty on all ```cpus``` affected by an address space change. You also need to instrument ```Pd::make_current``` to see when that mark is cleared and whether the TLB is flushed.

A log with those events would be helpful.

alex-ab, 2015-01-09:
Based on r8_genode_nosmp branch I applied following [kernel patch] (https://github.com/alex-ab/genode/blob/fa933b115c0fd11f1be567bc10b1530f21c65367/kernel-diff-2.txt) to get following [log output](https://github.com/alex-ab/genode/blob/fa933b115c0fd11f1be567bc10b1530f21c65367/trace-log-2.txt).

did 24 is the first VBox VMM protection domain - did 25 the protection domain of the first vcpu 
did 29 is the second VBox VMM protection domain - did 30 the protection domain of the second vcpu

alex-ab, 2015-01-12:
Some new infos. Just disabling the vpid feature but not the pcid feature also helps. And it seems to be a 64bit only problem (pcid are only supported for 64bit anyway, but the vpid feature also is there on 32bit). So a overview:

multiple Seoul VMs - 32bit Tinycore/Linux guest
------------------------------------------------------------
32bit host  cmdline "serial iommu"            - works
64bit host  cmdline "serial iommu"            - fails
64bit host  cmdline "serial iommu nopcid" - works
64bit host  cmdline "serial iommu novpid" - works

multiple VBox VMMs - 32bit Win7 guest
--------------------------------------------------
32bit host  cmdline "serial iommu"            - works
64bit host  cmdline "serial iommu"            - fails
64bit host  cmdline "serial iommu nopcid" - works
64bit host  cmdline "serial iommu novpid" - works

alex-ab, 2015-03-05:
@chelmuth, @nfeske: 1dd3dba57202a8169c1ee1dabdae288895cdbcda fixes the observed hanging system when multiple VBox VMs are running

alex-ab, 2015-03-27:
Can we close the issue ? Multiple VMs can now be executed concurrently. Still the kernel issue exists - when leaving out "novpid" - but this is another (non-Genode) issue.

chelmuth, 2015-03-27:
Ack.

closed 2015-03-27 by chelmuth