genodelabs/genode #1345: Tool for printing current contrib directory of port

chelmuth, 2015-01-07:
This tool is very helpful during development on branches using various versions of ports, e.g., when bisecting for an issue.

cnuke, 2015-01-07:
Fwiw, there is also dd41286089b3c51ef32402cdc57339f386ae8565.

chelmuth, 2015-01-07:
Hm, looks pretty similar to my solution except that I wasn't aware of *tool/ports/mk/check_port_arg.inc*.

alex-ab, 2015-01-12:
Just a question. I don't like to "pollute" my shell history with all kind of hashes - therefore I use a script very very similar to yours @chelmuth. It simply sets in contrib/<portname>-hash a symbolic link to contrib/<portname>, so that I can reuse my last commands when switching branches and potentially versions of ports. So, does it also qualify for addition ?

```
#!/usr/bin/make -f

#
# \brief  Tool for updating the hash file of a port
# \author Norman Feske
# \date   2014-05-07
#

#
# Determine Genode base directory based on the known location of the
# 'create_builddir' tool within the Genode source tree
#
GENODE_DIR := $(realpath $(dir $(MAKEFILE_LIST))/../..)

include $(GENODE_DIR)/tool/ports/mk/front_end.inc
include $(GENODE_DIR)/tool/ports/mk/check_port_arg.inc

usage:
	@$(ECHO)
	@$(ECHO) "--- create symbolic link from 'contrib/<port-name>-<hash>' to contrib/<port-name> ---"
	@$(ECHO) "usage: update_hash <port-name>"
	@$(ECHO)

$(TARGET):
	$(VERBOSE)ln -snf $(PORT_DIR) $(CONTRIB_DIR)/$(PORT_NAME)
```

nfeske, 2015-01-12:
Interesting. I have nothing against adding such snippets to tool/ports. It seems very convenient. But we have to give it a proper name. In my opinion, "update_link" does not cut it. What do you think about referring to your link as a "shortcut" (what it is, right?) and hence calling the script "update_shortcut" or simply "shortcut"? The notion of such a "shortcut" could then be described in the ports documentation.

Btw, in the brief description, "to of" should be "to".

chelmuth, 2015-01-12:
Now, we only have to decide which snippets should go into *tool/port*
- [ ] current_hash
- [ ] current
- [X] shortcut

nfeske, 2015-01-12:
True, there is a little overlap here ;-)

Personally, I would possibly like to use the "current" script. Getting a complete directory is quite handy.

cnuke, 2015-01-12:
Creating a symlink is something I do by using another tool. Therefore +1 for “current”.

chelmuth, 2015-01-12:
So, I'll go for current and shortcut if there are no objections.

commit 1a8bae9a (2015-01-14): Tools for convenient handling of port contrib dirs

tool/ports/shortcut

  create symbolic link from 'contrib/<port-name>-<hash>' to
  contrib/<port-name>

tool/ports/current

  print current contrib directory of port

Fixes #1345.
commit f99ca9e3 (2015-01-20): Tools for convenient handling of port contrib dirs

tool/ports/shortcut

  create symbolic link from 'contrib/<port-name>-<hash>' to
  contrib/<port-name>

tool/ports/current

  print current contrib directory of port

Fixes #1345.
closed 2015-01-20 by chelmuth