genodelabs/genode #1347: x86: Move crt0 to .text.crt0 section

nfeske, 2015-01-07:
On ARM, the crt0 code is located in the dedicated ".text.crt0" section. This way, ELF entrypoint always points to the very beginning of the text segment. This is useful when converting an ELF image a raw binary (which can then be started by meremly jumping to the start of the image).

Even though this mechanism is not used on x86, I would like to see this aspect of the ELF layout unified between the architectures so that the entrypoint points always to the beginnging of the text segment on all architectures.

chelmuth, 2015-01-08:
I implemented this feature for static binaries in 727ac4bc0cea535ad31d8fde2ad9411b3f5e2850. Do you have further requirements?

nfeske, 2015-01-08:
@chelmuth That's actually all I wanted. Thanks for the quick response to the issue.

closed 2015-01-08 by chelmuth

nfeske, 2015-01-08:
Funny coincidence. Apparently, I am a bit out of touch with master. ;-)