genodelabs/genode #1348: input_session: clients of servers can trigger uncaught exceptions

alex-ab, 2015-01-07:
I had the case the in liquid_fb a thread died because of an uncaught exception as soon as the ring buffer becomes full. In my case an unresponsive client  didn't dequeued quick enough the events and the thread inside the server died.

Beside liquid_fb there seem to be also following services effected:

repos/demo/src/server/liquid_framebuffer/services.cc
repos/os/src/server/input_merger/main.cc
repos/os/src/server/nit_fb/main.cc
repos/os/src/server/nitpicker/main.cc
repos/os/src/drivers/framebuffer/sdl/fb_sdl.cc
repos/gems/src/app/launcher/dialog_nitpicker.h
repos/gems/src/server/wm/decorator_nitpicker.h
repos/gems/src/server/wm/nitpicker.h

Only 

repos/dde_linux/src/lib/usb/input/input_component.cc

handles this case.

My question: Just caught the exception and drop the input event ? If yes - silently or with some warning ?

In my case the client recovered after a while and managed to read all events - so the warnings disappeared after a while. However if you have a client which is dead and process no events for sure anymore - the warnings could become annoying. Opinions ?

I tend to catch the exception for all named services above and print a warning if this happens ...

alex-ab, 2015-01-07:
Sorry, repos/gems/src/server/wm/nitpicker.h is not affected.

nfeske, 2015-01-07:
What do you think about applying the solution of ```wm/nitpicker.h```?

nfeske, 2015-01-07:
Or maybe better, what about moving the exception handing as done in ```wm/nitpicker.h``` to ```Input::Session_component::submit```? This would fix all components at once. Components that don't want to use the default handling of this condition could still call ```event_queue().avail_capacity()``` before submitting a new event.

alex-ab, 2015-01-08:
@nfeske: you mean that way ?

nfeske, 2015-01-09:
Almost. Your solution has the problem that once the queue overflows, the log gets flooded with messages, one message per discarded event.

Hence, the wm/nitpicker.h version resets the event queue in the exception case. I would move the diagnostic message along with the reset of the event queue to the exception handler in Input::Session_component.

With this in place, you can remove avail_capacity checks from the users of the Input::Session_component class. I'm afraid that my last comment was a bit ambiguous. I actually don't think that checking for avail_capacity makes much sense. Sooner or later, the input server needs to discard events anyway if no one is listening. But if, for whatever reason, the server wants to avoid the automatic event-queue reset by Input::Session_component, it could avoid it (by asking for the avail capacity before submitting an event).

nfeske, 2015-01-09:
Very nice!

commit 0f18ecc1 (2015-01-09): base: handle input overflow exception

Fixes #1348
closed 2015-01-12 by alex-ab