genodelabs/genode #1349: ldso: split main.cc into multiple headers

nfeske, 2015-01-08:
Right now, the code of the dynamic linker looks very clean and well structured but it is still hard to comprehend as most of the code is contained in a single file: https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/ldso/main.cc

To ease the review of the code by third parties, it would be beneficial to split the file into multiple headers - ideally one header per class. The header of each file should contain a brief description of the purpose of the class, maybe referencing certain sections of the ELF specification.

@ssumpf What do you think?

ssumpf, 2015-01-08:
Sure, we could, for example, separate the dynamic section code, the shared-object interface, the exception handling support,  and the ELF objects (libs, binary, linker).

nfeske, 2015-01-08:
Sounds good. I would like to leave the granularity up to you.

ssumpf, 2015-03-17:
@nfeske: First version, please review.

nfeske, 2015-03-18:
This looks superb! Thank you for reworking it. :-)

chelmuth, 2015-03-19:
Merged to master.

closed 2015-03-19 by chelmuth