genodelabs/genode #1350: ldso: ctors in shared libraries loaded for the 'Shared_object' API

nfeske, 2015-01-08:
When loading a shared library via the ```Shared_object``` interface, I expect that global constructors contained in the library are executed. However, in the current version of the dynamic linker, this does not seem to happen.

nfeske, 2015-01-08:
The commit https://github.com/nfeske/genode/commit/d62c2e3e9aef9b9435ce7bd4ab962fa642f03004 adds a corresponding test to ldso.run.

When executed (i.e., on base-linux), the test gets stuck at the construction of the ```Shared_object```.

ssumpf, 2015-01-08:
@nfeske: Ok, I will look into that.

ssumpf, 2015-01-09:
@nfeske, @chelmuth: Please cherry pick both commits.

nfeske, 2015-01-09:
Why does the change of the run script appears in the first commit? Shouldn't this be part of the second one (the test)?

chelmuth, 2015-01-09:
I merged both commits. Is that okay with you (both)?

ssumpf, 2015-01-09:
@nfeske: That is because the run script changes where missing in your commit. As @chelmuth suggested, I would just squash the two of them.

commit 4d8b2302 (2015-01-09): ldso: shared-object lock and Shared_object ctor test

This has to be used during shared object creation and destruction because global
lists are manipulated. We cannot use the 'Elf_object::lock' here because there
may be jump-slot relocations during object initialization.

Fixes #1350
commit 1a26f334 (2015-01-09): ldso: shared-object lock and ctor test

This has to be used during shared object creation and destruction because global
lists are manipulated. We cannot use the 'Elf_object::lock' here because there
may be jump-slot relocations during object initialization.

Fixes #1350
closed 2015-01-12 by ssumpf