genodelabs/genode #1351: hw vea9x4: remove trustzone support

m-stein, 2015-01-08:


m-stein, 2015-01-08:
Fixed by https://github.com/m-stein/genode/commit/9a4a0448bed1d71f7cf153632bc0b450c1214e14 that is based on staging.

Should fix recent builtbot problems with ``hw_vea9x4_tz`` ;)

chelmuth, 2015-01-08:
That's unexpected (at least)...

nfeske, 2015-01-08:
@skalk and @m-stein spoke with me about it earlier today. Right now, the vea9x4 TrustZone support is not in a state where we can run a large number of tests. To fix that, we'd require to invest a significant amount of time. But nobody is actually using TrustZone on this platform. So the work would not benefit anyone really. So we have two options:

* Keeping the TrustZone support for vea9x4, but untested (and thereby possibly broken)
* Removing it, keeping only things in base-hw that are thoroughly tested

I support the second option. Whenever someone needs to have TrustZone on the vea9x4 platform re-enabled, we still can revive the old code from our Git history.

commit 9a4a0448 (2015-01-08): hw vea9x4: remove trustzone support

When building Genode for VEA9X4 as micro-hypervisor protected by the ARM
TrustZone hardware we ran into limitations regarding our basic daily
testing routines. The most significant is that, when speaking about RAM
partitioning, the only available options are to configure the whole SRAM
to be secure and the whole DDR-RAM to be non-secure or vice versa. The
SRAM however provides only 32 MB which isn't enough for both a
representative non-secure guest OS or a secure Genode that is still
capable of passing our basic tests. This initiated our decision to
remove  [... cut]
closed 2015-01-12 by m-stein