genodelabs/genode #1355: modularise run tool [RFC]

cnuke, 2015-01-09:
The run tool served us well over the years and by continously extending its 
functionality it satisfied all our needs. The way it was extended up to now 
was in a ad-hoc fashion that makes future changes and maintaining more
time-consuming. In order to address these problems the run tool is getting
modularised and furthermore its argument handling is getting unified.

When executing a run script all tasks that needs to be performed, i.e.,
preparing the target system, creating a usable image and so, are now 
provided by specific backend modules. These modules are located in 'tool/run'.

As a preview the current version can by found at https://github.com/cnuke/genode/commits/modular_run.
More detailed information is available in 'tool/run/README'.

Since deploying the new run tool version includes adapting possibly all 
existing run scripts this task is post-poned until everyone is fine with
the new structure/API.

One remark that @nfeske had is splitting the 'reset' module into 'poweron'
and 'poweroff' — the current version still uses 'reset'. In addition to
that, maybe someone has other ideas how to structure the modules in a more
fine way?

nfeske, 2015-01-09:
Your solution is seriously cool! It is so.... clean! :-)

nfeske, 2015-01-10:
After thinking a bit more about the new run version, it occurred to me that, given the new degree of modularity, we could even get rid of the concept of run environments. After your restructuring, the remaining functionality of the run environments comes down to the assembly of the boot directory. So how about just modelling this as a run step? We could introduce a ```tool/run/boot_dir/``` directory with one file for each base platform.

nfeske, 2015-01-10:
As a (rather cosmetic) issue, I find the term "output" a bit too generic. I would prefer to rename "output" to "log" to draw the connection to core's LOG service. Would that be ok?

cnuke, 2015-01-10:
Well, we could do that but than we would have to add the `run_genode_until` hook at a different location.

Changing output to log is fine by me.

skalk, 2015-01-12:
I wouldn't replace 'reset' with 'poweron' and 'poweroff'. Althoug when reseting via AMT or network-controlled power plug we always do 'poweroff' && 'poweron' to reset, there is a new case where we use a switch connected to the reset jumpers of a PC. In that case, we really just want to do a 'reset'.

nfeske, 2015-01-12:
For this case, we could configure the buildbot's build directory to simply use a poweron/reset step that performs just that - a reset.

nfeske, 2015-01-12:
We would not use the run mechanism for powering-on the machine.

chelmuth, 2015-01-12:
```
genode_staging.git (modular_run)> ./tool/create_builddir nova_x86_32
Directory /plain/krishna/src/genode/genode_staging.git/build/nova_x86_32 already exists.
make: *** [build_dir_already_exists] Error 1
```

chelmuth, 2015-01-12:
Sorry, this is the error I had
```
genode_staging.git (modular_run)> ./tool/create_builddir nova_x86_32
cat: /plain/krishna/src/genode/genode_staging.git/tool/builddir/etc/build.conf.conf.run_opt_example: No such file or directory
make: *** [/plain/krishna/src/genode/genode_staging.git/build/nova_x86_32/etc/build.conf] Error 1
```

cnuke, 2015-01-12:
While moving <base>/run/env → tool/run/boot_dir it came to my attention that we have to deal we the handling of the kernel msg that is used by `log/serial`. What was the reason for checking it in expect in the first place?

All in all I am still not convinced that getting rid of the run env is really necessary…

chelmuth, 2015-01-12:
IMO the intention of the kernel message was to differentiate between simple boot errors and run-rime issues of the test itself.

cnuke, 2015-01-12:
Ah thanks, that makes sense. Now, where to put the kernel msg string, `boot_dir` is not the right choice and creating another type of module is likely unjustified.

cnuke, 2015-01-14:
Alright, I have incorporated all remarks so far (I did not squash the commits into a single one for now).

The new run tool is briefly tested by running a few simpler run scripts on various platforms (AMT/AMT + USB serial on nova and foc, hw_pbxa9 and linux).

To try out the new run tool, just check out https://github.com/cnuke/genode/commits/modular_run, create a new build directory, e.g. `tool/create_builddir nova_x86_64 BUILD_DIR=<path>/nova_x86_64_mod_run` and configure the `RUN_OPT` variable according to your setup. Reading `tool/run/README` and taking a look into the various modules in `tool/run` is recommended.

chelmuth, 2015-01-14:
:+1: I'll give it another shot then.

cnuke, 2015-01-19:
I cleaned up the commits and it is eventually ready for staging. We have to adapt the buildbot & co first, though.

chelmuth, 2015-01-20:
Please replace the multiple `Fixes #1355` annotations by just a reference to the issue on next branch update. Only the final commit should close the issue.

cnuke, 2015-01-20:
Okay, I will keep that in mind the next time.

cnuke, 2015-01-21:
@nfeske please merge the fixups to staging.

nfeske, 2015-01-21:
@cnuke done.

cnuke, 2015-01-22:
@nfeske please merge those two fixups to staging.

nfeske, 2015-01-22:
Done.

cnuke, 2015-01-23:
@nfeske or @chelmuth please merge to staging.

cnuke, 2015-01-23:
@nfeske could you please merge the last two commits 29e7285 and 2914f47 to staging.

nfeske, 2015-01-23:
I have merged the commits. Thanks for the reminder.

cnuke, 2015-01-24:
@nfeske or @chelmuth please merge both commits to staging.

chelmuth, 2015-01-24:
The commits just entered staging.

commit d2afb9e2 (2015-01-20): run: add power_on/softreset

The softreset module resets the target system by triggering a software
reset.

Fixes #1355.
commit 55bf0bb2 (2015-01-26): run: add power_on/softreset

The softreset module resets the target system by triggering a software
reset.

Fixes #1355.
closed 2015-01-26 by cnuke