genodelabs/genode #1359: device/PCI bus manager

cnuke, 2015-01-10:
So far, all drivers probe the entire PCI bus and attach to the first device
that is found and which the driver can manage. Therefore, it is not possible
to e.g. use two different NIC devices that are handled by the same driver
even when starting the driver twice at the same time.

One possible interim fix could be to configure the driver to only attach to a
particular device by specifying the vendor/device id via the <config> node.

The long-term solution on the other hand would probably be to limit the list
of devices on the PCI bus in the platform driver. Thereby a driver would only
see the device it is supposed to see. To take this idea further, a potential
device manager component could query its device/driver database and would
configure the PCI session accordingly and would start the driver.

nfeske, 2015-01-10:
As a first step, we should really establish the notion of a "platform driver" on x86 as we already do on ARM. We could start by renaming the ACPI driver to platform driver and renaming the ```Pci::Session``` interface to ```Platform::Session```. By doing that, we should make sure that the platform driver is functional on non-APIC kernels (OKL4, L4/Fiasco). Right now, the ACPI driver is incompatible with those kernels.

Once, we have the notion of a platform driver, let us change the way of how device drivers obtain their resources. Instead of using the platform driver to merely probe for device resources and then opening IO_MEM/IO_PORT/IRQ sessions at core, device resources should be obtained from the platform driver, more precisely from the respective ```Pci_device``` interface. This step will solve two problems:

* Drivers can no longer open sessions for arbitrary device resources at core but only the resources of a device the driver has a (device) capability for. The handing out of device capabilities will depend on the policy of the platform driver.
* Users will no longer accidentally bypass the IRQ remapping handling of the ACPI driver by routing IRQ sessions of drivers directly to core. Instead, all drivers will open a "Platform" session. There is no ambiguity anymore.

chelmuth, 2015-01-10:
Your proposals give us a very good starting point to finally cleanup the device handling. I already have a question about the session to the platform driver: May it suffice to provide only one device per session or do we still need the notion of a (virtual) bus incl. interfaces for iterating over devices on the bus?

nfeske, 2015-01-10:
After thinking a bit more about the new run version, it occurred to me that, given the new degree of modularity, we could even get rid of the concept of run environments. After your restructuring, the remaining functionality of the run environments comes down to the assembly of the boot directory. So how about just modelling this as a run step? We could introduce a ```tool/run/boot_dir/``` directory with one file for each base platform.

nfeske, 2015-01-10:
As a (rather cosmetic) issue, I find the term "output" a bit too generic. I would prefer to rename "output" to "log" to draw the connection to core's LOG service. Would that be ok?

nfeske, 2015-01-10:
Oh, wrong issue. Sorry.

nfeske, 2015-01-10:
I would keep the platform-session interface separate from the device interface - at least for now. I expect that the platform-session interface will certainly contain things that are not merely about handing out device resources. For example, on the Raspberry Pi, the platform driver contains the interface to the Videocore.

nfeske, 2015-01-10:
On the other hand, the Videocore access could be handled by introducing a dedicated "Videocore" session interface, which the Rpi platform driver could provide besides the "Device" service. That might be even cleaner. Hmm... So maybe the idea to completely disregard the bus concept is not too far off. I would probably still stick to it at first - just to prevent too many architectural changes at once.

chelmuth, 2015-01-10:
Agreed.

nfeske, 2015-07-22:
I propose to close this issue. What are your thoughts?

chelmuth, 2015-08-21:
I'd close the issue if @skalk and @alex-ab confirm that there are no missing pieces regarding ARM platforms.

alex-ab, 2015-08-21:
I can't confirm that for ARM platforms - we didn't apply this to the ARM platforms yet. Nevertheless I'm up for close.

chelmuth, 2016-01-07:
No objections raised -> closed.

closed 2016-01-07 by chelmuth