genodelabs/genode #1362: AHCI doesn't work on (some) real hardware

NobodyIII, 2015-01-13:
This is the problem I've talked about on the mailing list. Basically, AHCI doesn't work with my SSD. In my latest Genode scenario I've tested, I used a read-only version of test-blk-cli. The test gave no output. Here is the serial output from Genode:

Hello World.
Need 015d8000 bytes to relocate modules.
Relocating to bcb39000: 
Copying 260616 bytes...
Copying 10485760 bytes...
Copying 296724 bytes...
Copying 900576 bytes...
Copying 351204 bytes...
Copying 325488 bytes...
Copying 266228 bytes...
Copying 434196 bytes...
Copying 320292 bytes...
Copying 3514682 bytes...
Copying 463984 bytes...
Copying 136304 bytes...
Copying 1668924 bytes...
Copying 1079180 bytes...
Copying 296448 bytes...
Copying 297784 bytes...
Copying 263552 bytes...
Copying 313332 bytes...
Copying 1618 bytes...
Copying 483180 bytes...
Copying 242500 bytes...
Copying 392848 bytes...
Copying 56226 bytes...

Welcome to Fiasco(ia32)!
DD-L4(v2)/ia32 microkernel (C) 1998-2009 TU Dresden
Rev: rUNKNOWN compiled with gcc 4.7.2 for Intel Pentium    [] 
Performance-critical config option(s) detected:
  CONFIG_ASSEMBLER_IPC_SHORTCUT is off
  CONFIG_NDEBUG is off
  CONFIG_NO_FRAME_PTR is off
  CONFIG_FINE_GRAINED_CPUTIME is on

Enabling special fully nested mode for PIC
Using the PIT (i8254) on IRQ 0 for scheduling
SERIAL ESC: Using serial hack in slow timer handler.
SERIAL ESC: No IRQ for specified uart port.
Using serial hack in slow timer handler.
Absolute KIP Syscalls using: Sysenter
CPU: AuthenticAMD (15:2:0:0) Model: Unknown CPU at 3492 MHz

  48/512 Entry I TLB (4K pages)     24/512 Entry I TLB (4M pages)
  64/1024 Entry D TLB (4K pages)     64/1024 Entry D TLB (4M pages)
  64 KB L1 I Cache (2-way associative, 64 bytes per line)
  16 KB L1 D Cache (4-way associative, 64 bytes per line)
2048 KB L2 U Cache (8-way associative, 64 bytes per line)

Freeing init code/data: 20480 bytes (5 pages)

Calibrating timer loop... done.
SIGMA0: Hello!
  KIP @ 1000
  Found Fiasco: KIP syscalls: yes
  allocated 4KB for maintenance structures

KIP @ 1000
    magic: 4be6344c
  version: 87004444
         sigma0  esp: 002da72c  eip: 00104558
         sigma1  esp: 00000000  eip: 00000000
           root  esp: 00000000  eip: 010536ec
MBI @ 68000
 mod[3] [1212000,1212652) config
 mod[4] [1213000,125f7f4) init
 mod[5] [1260000,12a0580) timer
[... 145 lines cut ...]
[init -> acpi] MCFG BASE 0xe0000000 seg 00 bus 00-ff
[init -> acpi -> pci_drv] PCI driver started
[init -> acpi -> pci_drv] Could not open file "pci_device_pd"
[init -> acpi -> pci_drv] PCI device protection domain for IOMMU support is note
[init -> acpi -> pci_drv] 00 BDF start 0, functions: 0x10000, physical base 0xe0
[init -> acpi] ACPI table format is supported - rewrite GSIs
[init -> acpi] Rewriting 01:00.0 IRQ: 10 -> GSI: 24
[init -> acpi] Rewriting 01:00.1 IRQ: 11 -> GSI: 25
[init -> acpi] Rewriting 02:00.0 IRQ: 10 -> GSI: 48
[init -> acpi] Rewriting 03:05.0 IRQ: 10 -> GSI: 20
[init -> acpi] Rewriting 05:00.0 IRQ: 11 -> GSI: 19
[init] child "acpi" announces service "PCI"
[init] child "acpi" announces service "IRQ"
[init -> ahci] Found AHCI HBA (Vendor ID: 1002 Device ID: 4391 Class: 00010601)
[init -> ahci] Port 0: ATAPI no
[init -> ahci] Port 0: Detected interface is active
[init -> ahci] AHCI Version: 1.0200
[init -> ahci] CAPs:
[init -> ahci]  Port count: 4
[init -> ahci]  Command slots: 32
[init -> ahci]  AHCI only: yes
[init -> ahci]  Native command queuing: yes
[init -> ahci]  64 Bit: yes

Note that the scenario works perfectly with rom_blk, so configuration isn't the issue.

skalk, 2015-01-13:
When looking at your log output, I can see that you start rom_blk as well as ahci. Both components announces a block service. Would you please provide your run script too?

NobodyIII, 2015-01-13:
I can't right now, but ROM_blk is there so I can easily switch between it
and AHCI. My configuration should not be the issue.

skalk, 2015-01-14:
It would be nice anyway to take exactly your configuration to reproduce your results. Please post it as soon as you have access to it again. Thanks in advance.

NobodyIII, 2015-01-14:
Here it is.

chelmuth, 2015-01-15:
@NobodyIII the files did not make it to Github. I suggest you open a https://gist.github.com/ and post the link in this thread.

NobodyIII, 2015-01-15:
Here is a Google Drive link.
   config
<https://docs.google.com/file/d/0Byr3WX_C3550WGlXZ2doUDNVWGtZQnFwWFVUTHBQaDVZZS00/edit?usp=drivesdk>

chelmuth, 2015-05-27:
This issue applies to the old AHCI driver. Reopen for the new driver if the issue persists.

closed 2015-05-27 by chelmuth