genodelabs/genode #1372: Separate known-to-work from experimental run scripts

nfeske, 2015-01-16:
Currently, there exist approximately 200 run scripts scattered over the various ```repos/<repo>/run/``` directories. Reportedly, the number and varying quality of the run scripts is cumbersome to newcomers. There exist different kinds of run scripts:

* The 47 run scripts listed in ```tool/autopilot.list``` are executed each night on various base platforms. Genererally, they are known to work and they are able to detect their successful completion automatically.
* There are run scripts created as a side effect while developing a particular component. They are used as a mere work-flow tool and are not meant to be generally usable. Often, they are tied to the particular base platform the developer is using while developing. For the most part, those run scripts lack exit conditions but just run infinitely. So they are not suitable as test cases as they are.
* There are integrated demonstration scenarios. Those are rarely executed and revived only from time to time. They are nice because they demonstrate how the run mechanism can be used to build sophisticated systems but they are not suitable as a basis for learning. More often than not, they are tied to a particular hardware/kernel combination. Consequently, we have largely removed these kind of run scripts from the repository. They still reside on various branches.

The second category represents by far the majority of run scripts. Because those are not tested on a regular basis but only when a developer picks them up occasionally, they often do not work out of the box when newcomers try them out. This leaves (understandably) a bad aftertaste.

I propose two steps to improve the situation:

1. Separate the known-to-work run scripts of the first category from the others. All run scripts not covered in ```autopilot.list``` should go to a distinct place, i.e., ```repos/attic/run/```. The "attic" repository must be explicitly enabled in the build-directory's ```build.conf```. The run scripts found there are _documented to be shaky_. So people who use them know that they are walking on shaky ground. All run scripts found in the other repositories are regularly (automatically) tested, which assures a smooth experience when exploring Genode. Because the run scripts listed in autopilot.list correspond to the run scripts of the non-attic repositories, we can remove the autopilot.list file. The buildbot could simply grep all run scripts from the repositories, except those from the attic.
2. Iteratively move run scripts from the attic to the other repositories and thereby improve the test coverage of components. The major activity here will be to introduce proper exit conditions into the run scripts and generalize them so that they will work across many base platforms. Any base platform that is known to not work must be captured by an assertion at the beginning of the run script.

For the second part, we need a way to distinguish the interactive execution from the automated execution. E.g., when executing a graphical scenario interactively, we don't want it to automatically stop at some point. But when executing the same run script by the autopilot, it should detect a success and exit.

cnuke, 2015-01-16:
:+1: On that account, I would adapt the run scripts executed by the autopilot to work with the new run tool. The remaining run scripts can than be adapted on demand.

nfeske, 2015-01-16:
@cnuke Good call!

cnuke, 2015-01-21:
All non-regularly tested run scripts are moved to `repos/attic/run`. Each script is prefixed with the name of the repository it came from (that should make it easier if its eventually moved back later).
The next point on the to-do list would be adapting the documentation.

nfeske, 2015-01-22:
@cnuke Great that you went ahead with the creation of the attic. But please let us keep the original run-script names. Adding a prefix like you did has two disadvantages:

1. I breaks all documentation and mailing list postings where we gave an instruction like ```make run/<run-script-name>```. Moving a run script back to its original repository would break it again.  This is really bad.

2. The name is part of the user interface of the build system. It is taken as argument to ```make run/```, which the user has to manually type in. The prefix makes the user interface much more clunky. Instead of typing ```make run/eglgears```, one would have to type ```make run/libports_eglgears```. That is really unwieldy.

The only added feature that you cited in favour of the prefix - to let it implicitly document the origin of the run script - can be achieved quite easily by a look into the Git history.

In the rare cases where run script names are ambiguous, let us disambiguate them by adding a short prefix. E.g., the base-hw-specific run script "cpu_scheduler.run" should be renamed to "hw_cpu_scheduler.run". For Linux-specific run scripts, this is already the case as they are prefixed with "lx_".

cnuke, 2015-01-22:
Fine by me.

skalk, 2016-06-28:
What about this issue? It seems everybody was fine with the intention, but it did not evolve further. If I remember correctly, the problem was that certain interactive, non-automatic tests should not be moved to attic, strictly speaking the demo.run script. But even if we leave out this file(s), why didn't we moved the other run-scripts anyway?
From my perspective, I would even vote for removing non-maintained scripts at all instead of introducing an "attic" repository. Run-scripts that are not used in nightly tests or maintained accordingly, can also reside in topic branches of the developers. It would be more motivation to add automated tests to new components, if you want your run-script to reside in the mainline branches.

Anyway, my intention to reanimate this discussion is the current restructuring of our nightly buildbot. Again I stumbled over the "autopilot.list" file. IMO it would be better to see straightaway whether a run-script is an automatic one, or an interactive one. Maybe we could prefix the "autopilot-capable" run-scripts from the other ones by prefixing them, e.g., `auto_lwip.run`? This would be helpful for newbies too. There were several questions in the past, when people tried to interact with automated scenarios, e.g., `tz_vmm.run`.

nfeske, 2016-06-28:
In the context of @cproc's work on automating the USB HID test, we came up with the idea to allow a run script to be usable in both an interactive and automated way by evaluating an argument `---auto` passed to the run tool. The autopilot would always specify this argument (hence executing the run script in an automated way) whereas a freshly created build directory would run the interactive version unless `--auto` is added to `RUN_OPT`. I think that this approach could be applied to some of the current interactive run scripts as well. But I am afraid that not all of them can be easily covered.

I am particularly puzzled about the ones at gems/run/. On the one hand, turning them to automatically executable run scripts is quite difficult or would at least spoil the simplicity of the examples. On the other hand, they serve as a very effective kind of documentation. Moving them out of sight to a topic branch buries this valuable information. Even in the event that one of those run scripts break, it still embodies the blue print for using a particular component. In most cases when run scripts break, it is for quite mundane problems such as a lack of quota, or a missing boot module - problems that are unrelated to the actual component at hand. Even if a run script is not working flawlessly, it is still not worthless altogether.

skalk, 2016-06-28:
Thanks for the insights.
I suppose, the `--auto` option will return in in a common way when run-scripts do not support automated testing. So that we get rid of `autopilot.list`, and instead execute `repos/*/run/*.run` with the `--auto` option in future?

nfeske, 2016-06-28:
I haven't thought that far. But I fail to see the appeal of removing the `autopilot.list`. What is the problem with it?

skalk, 2016-06-29:
My intention was not to care for this file when enabling automated tests, but nevermind it is not important.

nfeske, 2020-02-28:
With the introduction of the depot autopilot, this issue became much less of a burden. I'm closing it because I don't expect any further activity in this direction.

closed 2020-02-28 by nfeske