genodelabs/genode #1373: Init: support aliases for child names

nfeske, 2015-01-16:
It would be nice to be able unify init configurations across different platforms.

Right now, when declaring an explicit route to a ```fb_drv``` server in a run script, the run script won't work on Linux because on this platform, the framebuffer driver is called ```fb_sdl```. If ```fb_drv``` had a way to also appear under the name ```fb_drv```, the route could be resolved.

Another example is the role of the USB driver. Depending on the platform, the USB driver is an input driver, a block driver, a networking driver, or a combination of those. When declaring routes of components using one of those services, it would be great to specify a route to ```input_drv``` or ```nic_drv``` instead of routing sessions to ```usb_drv```.

In short, the support for aliases has a great potential to unify/simplify run scripts.

chelmuth, 2015-01-19:
The feature overlaps somewhat with the combination of `<start>` and `<binary>` as we we use it currently in some scripts. But regrettably I can't come up with a clear solution to prevent the overlap.

nfeske, 2015-01-19:
I fail to see how this feature overlaps with the ```<binary>``` sub node. The ```<binary>``` sub node is barely a feature. It defines how the binary of a component is called. The support for omitting the ```<binary>``` in the special case where the binary name equals the component name may be considered as a feature. But how does it overlap with the ```<alias>``` that I proposed? An alias allows one child to appear as multiple different names when sessions are routed. It is not an attribute of the child. To give an example:

```
<start name="fb_sdl">
  ...
</start>
<alias name="fb_drv" child="fb_sdl"/>
```

Commit https://github.com/genodelabs/genode/commit/e033fe8e9609feff49ecb1a2048c27584d826b27 adds support for such ```<alias>``` nodes to init.

chelmuth, 2015-01-19:
Thanks for the clarification.

commit e033fe8e (2015-01-19): Init: support for aliases for child names

Fixes #1373
closed 2015-01-20 by nfeske