genodelabs/genode #1380: tool_chain_rump build fails with GCC 4.8.1

nfeske, 2015-01-21:
On my Ubuntu  13.10 system, ```./tool/tool_chain_rump build``` fails with the following error:

```
    compile  rump_allserver/rump_allserver.o
/open/contrib/rumprun/rumpsrc/usr.bin/rump_server/../rump_allserver/rump_allserver.c: In function 'main':
/open/contrib/rumprun/rumpsrc/usr.bin/rump_allserver/rump_allserver.c: In function 'main':
/open/contrib/rumprun/rumpsrc/usr.bin/rump_allserver/rump_allserver.c:455:4: error: overflow in implicit constant conversion [-Werror=overflow]
    flen = RUMP_ETFS_SIZE_ENDOFF;
    ^
/open/contrib/rumprun/rumpsrc/usr.bin/rump_server/../rump_allserver/rump_allserver.c:455:4: error: overflow in implicit constant conversion [-Werror=overflow]
    flen = RUMP_ETFS_SIZE_ENDOFF;
    ^
```

The GCC version is ```gcc (Ubuntu/Linaro 4.8.1-10ubuntu9) 4.8.1```.

chelmuth, 2015-01-21:
It may help to `apt-get install gcc-4.7` but I did not try `CC=gcc-4.7 tool/tool_chain_rump` myself yet.

nfeske, 2015-01-22:
No luck with specifying gcc-4.7:

```
building rump tools...
>>
>> WARNING: repository in /open/contrib/rumprun/rumpsrc is not clean
>>
>> ERROR:
>> Need GNU or BSD ar (i686-linux-gnu-ar: not found)
make: *** [contrib/rumprun/.compile.tag] Error 1
```

cnuke, 2015-01-22:
This assignment misses a cast, I'll prepare a patch.

nfeske, 2015-01-26:
Thanks. After your update, it works.

closed 2015-01-26 by nfeske