genodelabs/genode #1381: vfs / rtc_file_system: remove dependency from libc

nfeske, 2015-01-22:
The ```Rtc_file_system``` includes the libc header ```time.h``` in order to call ```localtime()```. Unfortunately, this convenience shortcut makes the VFS depend on the libc. It eventually prevents us from moving the VFS into a stand-alone library to be used by libc-less programs. To solve the issue, I see two approaches:

1. We could host the code for the time conversion within the ```Rtc_file_system```.

2. We could change the ```Rtc::Session``` interface to use a structured type for representing time. This would alleviate the need to perform any bit-fiddling at the client (i.e., the RTC file system).

I would prefer the second option.

cnuke, 2015-01-22:
:+1: for the second option.

tomga, 2015-01-25:
I don't know why this call to `localtime()` is needed but second option seems to force all users of generic api `Rtc::Session` to use single time zone - local. Or maybe proposed change suggests to only add some time conversion functions to provide similiar functionality to provided right now by libc.

I don't know if I can find time but I would investigate why `localtime()` is needed at all in vfs as it doesn't seem to be obvious at all. Conversion to time in concrete time zone seems to be needed for presentation to end users not in some generic apis.

cnuke, 2015-01-26:
Well, presentation is precisely the point of `/dev/rtc`. It should provide the user with the current date in human readable form. Using `localtime()` at this point was merely a convenient but arguably a bad call to convert the UNIX timestamp given by the Rtc_session to a `struct tm`.

The proposed change of API involves replacing the method giving the UNIX timestamp with a method that returns a more structured value. In any case, it should **always** return the time in UTC¹. Any actual conversion has to be made at the client.

¹) I personally would not care about systems using local for its RTC — it could be made configurable, though.

chelmuth, 2015-01-26:
I'd also like to emphasize that the RTC service will provide the time in UTC. For me it makes no sense to configure the hardware RTC in local time since even MS Windows supports RTC in UTC.

tomga, 2015-01-26:
I haven't been considering RTC in non UTC. It's just I've always thought about `time_t` as representing concrete time in timezone independent way and `struct tm` as timezone aware structure.

Given that I expect that client application to get presentation in local time (or some other time zone) would have to convert received `UTC tm` to `time_t` and then to `local tm`. Maybe this way of thinking is caused mostly by being accustomed to `localtime()` and `gmtime()` typically used for time conversions.

Anyway it seems to me that you are proposing to change api due to wrong reasons. If you need to convert `time_t` to presentation in UTC without dependency to `libc` it seems to be trivial to do that locally in just few lines of code (`localtime()` processing is a different beast). And additionally any time processing is simpler using `time_t` than using `struct tm`. To count periods of time client will have to convert to `time_t` anyway because it is just easier. And of course `time_t` is smaller in terms of space.

I will not try more to convince you it is a bad idea. Decision is up to you.

chelmuth, 2015-01-27:
@tomga depicts what I also thought about the limitation to a structured timestamp value only. Therefore, I propose to provide RTC like follows.
``` c
/* Time stamp in UTC */
struct
{
	unsigned         microsecond;
	unsigned         second;
	unsigned         minute;
	unsigned         hour;
	unsigned         day;
	unsigned         month;
	unsigned         year;
	Genode::uint64_t epoch_time; /* microseconds since Epoch 1970-01-01 00:00:00 */
};
```

nfeske, 2015-01-27:
@chelmuth Isn't ```epoch_time``` redundant? It can be calculated from the other values, doesn't it?

Apart from that, such a structure is exactly what I had in mind.

chelmuth, 2015-01-27:
@nfeske you're right: For the calculation of epoch_time from the values above the reverse of computation of localtime() is needed. This function is currently implemented in RTC. Would it be an option to provide this utility?

cnuke, 2015-01-27:
The current pending implementation already uses the structure proposed by @chelmuth minus the `epoch_time` field. For all clients using libc this API change is insvisible. For all other clients, the only change is, that a `mktime()`-like function has to be executed locally if a `time_t`-like value is needed. If you really need more functionality than that, using the libc is currently your only option on Genode anyway and you do not care which structure is used to query the RTC. Therefore, you also do not care that the structure transporting the information uses more space than `time_t`. On the other hand, it becomes easier for clients that do not use the libc to provide a readable format, e.g., in LOG output.

tl;dr we provide the real RTC value (in UTC) in a more structured format than `time_t` for convenience sake and clients are free to convert this format to any structure and any timezone they like.

nfeske, 2015-01-27:
@cnuke Please leave out the notion of UTC in the RTC session interface. At the session-interface level, the timezone should not be a concern. E.g., whereas an RTC driver might produce a UTC time, a local_rtc server may be used to transform the UTC time reported by the driver to a local time and, in turn, offer this local time as another RTC service. Since both services would provide the same RTC session interface, it does not make sense to tie the session interface to UTC.

cnuke, 2015-01-27:
Yeah, sorry about mentioning that. There is obviously no notion of UTC at the RTC session interface.

tomga, 2015-01-27:
So you were choosing which clients of `RTC` clients are forced to use `libc` as a dependency. Currently you are considering usage in logs so you've chosen a structured variant. If later you will somewhere need to quickly count execution time of some operations you will be forced to use functions from `libc` to convert to `time_t` like value to be able to subtract values.

Maybe it is really better (although I don't think so). I think that decision is made due to wrong reasons - to remove `libc` dependency on ONE client service - not taking into account other usages on which you force usinng `libc` as a dependency. Anyway it is not that important to fight for it more.

nfeske, 2015-01-27:
@tomga I do not understand this controversy at all. Please consider the following:

* You can have a separate component that is both an RTC client and an RTC server. Such a component can perform arbitrary transformations (like UTC to local time). There can even be an arbitrary number of such components present, each presenting a differently transformed time.

* RTC clients are not forced to use the libc at all. If a client opens an RTC connection, it uses the time provided by the RTC server instance it got routed to. The session routing is just a configuration.

* RTC requests are rare and slow. There is no point on micro-optimizing the data format used for the transport of the information from the RTC server to the client. For fine-grained performance measurements, RTC is not the right tool.

* The extraction of useful information from a time_t value is more complex than accessing a member of a struct. The solution avoids this complexity for the cases that we have and foresee.

To me, the discussion smells like bike shedding.

tomga, 2015-01-27:
@nfeske, you wrote: "RTC requests are rare" and that was the information I was missing. I assumed that it can be used quite extensively - due to number of clients - almost every process. Thanks for explanation.

chelmuth, 2015-01-27:
@cnuke I'd like to merge your changes to staging, but do you agree we should squash all commits into one? The intermediate steps will not compile not to mention work properly, right?

cnuke, 2015-01-27:
@chelmuth yeah sure, it's fine by me.

commit 791dccbe (2015-01-28): os: structured timestamp in Rtc session

Instead of returning an uint64_t value, return a structured time stamp.
This change is only visible to components using Rtc_session directly.

Fixes #1381.
commit 0a835e4c (2015-02-16): os: structured timestamp in Rtc session

Instead of returning an uint64_t value, return a structured time stamp.
This change is only visible to components using Rtc_session directly.

Fixes #1381.
closed 2015-02-17 by cnuke