genodelabs/genode #1388: File system node locking

ehmry, 2015-01-30:
Ported programs will inevitably need to lock files to prevent race conditions between processes.

A simple way to do it would be lock files (the file x is locked when the file x.lock is present), but without access to global process information I think would be difficult to identify stale locks.

A server that stores lock information and proxies File_system service to another File_system server seems reasonable to me. It would add some overhead but I would think it could be applied conservatively through a VFS policy.

nfeske, 2015-02-04:
I like your proxy idea. This way, the file-system session interface could probably stay clean of the locking feature.

ehmry, 2015-06-25:
If I need it, I'll implement it.

closed 2015-06-25 by ehmry