genodelabs/genode #1390: foc: l4android doesn't compile nor run

skalk, 2015-02-02:
Currently, the compiler argues about the Genode ballooning driver, which was developed using l4linux and a more recent version of Linux. Moreover, it bleats about a cxx library related issue that was already fixed for L4Linux with commit eab8217bdb9f186d94840d038d915f2dd7d473ca

li94575, 2015-02-02:
I have replaced the funtion "l4re_ma_alloc" with previous version, namely, all memory is allocated once. and that error has gone, I don't know if it will affect other parts.

skalk, 2015-02-03:
@li94575 would you try out above commit 2858242 ? You will need to prepare l4android once more, I've moved our patches to a git clone of l4android.org.

li94575, 2015-02-03:
OK, I'll try out above commit, and prepare l4android again.

czzcz, 2015-02-04:
When prepared L4android ,  I got a fatal error: reference is not a tree...
I guess there is something wrong with your git url.

skalk, 2015-02-04:
sorry, I've missed one update of the tree, now it's commit 60e5ec532d0094fc8e9906b5ddbd5e066a8143e9

li94575, 2015-02-05:
@skalk I have tested the new l4android with your patches, that error did not appear. Thank you!  Can you tell me which files have been changed in l4android?

commit 47c76010 (2015-02-03): L4Linux: make ballooning driver configureable

To circumvent compilation errors with the older L4Android Linux kernel
version, the ballooning driver is included in the more recent L4Linux
kernel only. Moreover, to be able to maintain L4Android / L4Linux in a more
convenient way, e.g. to apply patches valid for both versions, we use
the same git clone that is used for L4Linux instead of using the upstream
L4Android version by applying patches.

Fixes #1390
commit 0ac039fa (2015-02-16): L4Linux: make ballooning driver configureable

To circumvent compilation errors with the older L4Android Linux kernel
version, the ballooning driver is included in the more recent L4Linux
kernel only. Moreover, to be able to maintain L4Android / L4Linux in a more
convenient way, e.g. to apply patches valid for both versions, we use
the same git clone that is used for L4Linux instead of using the upstream
L4Android version by applying patches.

Fixes #1390
closed 2015-02-17 by skalk