genodelabs/genode #1394: nova: support usage of just one pager for all threads of one cpu

alex-ab, 2015-02-04:
On Genode/Nova we have currenlty one pager thread in core per each client thread in the system. We are forced to do so, since in core not every page fault can be resolved immediately - instead it get resolved later in time by another component. For this time period the pager thread must be blocked and can't be used by other threads.

Instead of blocking the thread in userland it would be useful to tell the kernel that the page fault is not handled immediatley and that the faulting thread is blocked inside the kernel. Extending the reply syscall by an optional semaphore cap would enable to block the faulting thread inside the semaphore by the kernel. 

With such a kernel change we can simplify core for Genode/Nova and converge closer to the implementation of other Genode base platforms. Additionally, with the change we can't run anymore out of threads. The overall number of threads in core would be static for a booted system - mainly dominated by the number of cpus and number of irqs.

alex-ab, 2015-02-04:
Seems to work. I need some more time for testing some advanced features like single-step and some stress testing (bomb, noux_tool_chain), but maybe some may have a look. There are some parts which look ugly and I open for useful suggestions to change that ;-).

nfeske, 2015-02-09:
Could you assist me with spotting the ugly parts ;-)

The only part that I do not like is the repetitive code for the exception handlers. I am wondering, do we really need all of them? If yes, couldn't we host them as ```Exception_handler``` object members of the ```Pager_object```? The portal creation could be done in the constructor of the ```Exception_handler``` and the handler function could be a method of this class. What do you think?

Are you planning to move the kernel patch to the NOVA Git repository eventually?

As a minor cosmetic detail, we are usually putting braces of functions at a new line. The patch does not always follow this style.

alex-ab, 2015-02-09:
> The only part that I do not like is the repetitive code for the exception handlers. I am wondering, do we really need all of them? If yes, couldn't we host them as Exception_handler object members [...]
We need them actually only for debugging reasons. So based on the exit you may identify the reason what went wrong. Gdb could be the only user actually and we, the human. 
I tend towards 'yes'. I try to make it a bit nicer.

> Are you planning to move the kernel patch to the NOVA Git repository eventually?
Yes.

> Could you assist me with spotting the ugly parts ;-)
One which is left is that I start several Pager_activations threads (one per CPU) under the hood of core. The general interface looks like that just one is running, but actually it are -n- pager activations.

commit 8b41c322 (2015-02-17): nova: use in core one pager per CPU

Fixes #1394
commit a3137847 (2015-02-27): nova: use in core one pager per CPU

Fixes #1394
closed 2015-02-27 by alex-ab