genodelabs/genode #1398: libports: sqlite

ehmry, 2015-02-04:
> SQLite is a software library that implements a self-contained, serverless, zero-configuration,
> transactional SQL database engine. SQLite is the most widely deployed SQL database engine
> in the world. - http://sqlite.org/

SQLite has an OS abstraction layer that handles interaction with filesystems and the system time, so a robust port can be made that is ignorant of SQLite's core.
http://sqlite.org/vfs.html

In order to share databases between processes, file locking is required.
https://github.com/genodelabs/genode/issues/1388
http://sqlite.org/lockingv3.html
http://sqlite.org/wal.html

I have a working SQLite VFS that wraps the libc VFS, I would like to migrate this to a native file_system session and then possibly implement the optional memory mapped I/O functions using filesystem packets.
http://sqlite.org/mmap.html

ehmry, 2015-02-24:
About 1/4 to 1/3 faster without libc VFS

nfeske, 2015-02-25:
@ehmry That is an impressive improvement. Now I just ask myself where all the time goes when using the libc VFS.

ehmry, 2015-02-25:
For what its worth the test isn't a realistic workload, but a lot of tiny writes from what I can tell.

ehmry, 2015-05-01:
Now there is a `sqlite` and `libc_sqlite`, and `run/sqlite_speedtest` and `run/sqlite_libc_speedtest` for comparison.

ehmry, 2015-08-28:
I am closing this issue because I have sqlite library in an external repo.

closed 2015-08-28 by ehmry