genodelabs/genode #1399: ports: nix

ehmry, 2015-02-04:
> Nix is a powerful package manager for Linux and other Unix systems that makes package
> management reliable and reproducible. It provides atomic upgrades and rollbacks, side-by-side
> installation of multiple versions of a package, multi-user package management and easy setup of
> build environments. - http://nixos.org/nix/

That is a glossy overview, the real details are here: http://dspace.library.uu.nl/handle/1874/7540.

Nix would be useful for building Noux packages and managing Noux environments, and possibly building and managing native components.

I would break the internals of Nix into three categories, inputs and expression evaulation, output building, and outputs managment. Input and expression evaulation is not difficult, and simple evaulation has been show to work. Building and output management however is very Unix specific, and much of its complexity is related to mitigating a departure from the Filesystem Hierarchy Standard.
A port is not trivial because a much of the current code would be redundant with the features of Noux and the stacked VFS, and I think this code would be more difficult to support than replace.

Dependencies are not much of an issue, just SQLite and some compression and crypto libraries.
https://github.com/genodelabs/genode/issues/1398

Some file locking will probably be required to prevent build races. On NixOS, the Linux distribution where Nix is most integrated, building is done with dedicated daemons that seem to communicate with locks. We could perhaps replace that with a Nix service were clients donate the resources for builds and let that service perform all the synchronization. Either way, it is a requirement that write access to the file system used for storing outputs be carefully restricted.
https://github.com/genodelabs/genode/issues/1388

johnku1, 2015-02-28:
What exactly needs to be done for this?

ehmry, 2015-02-28:
@johnku1 I'm not porting to Noux, I'm doing it native. Nix is broken down into a few sub-libraries, and the only one that needs much work is libnixstore. Libnixstore drives the database, manages locks, and handles the builder children. I'm rewriting some of that library now because I want native process control over the builders. Noux will sit between Nix and the Unix build processes. I was able to get that to work without too much effort, but it falls apart when the builders exit and Nix doesn't notice because it was using some Unix hackery to monitor them.

I got slowed down because I was trying to abstract the Unix stuff so both the Unix and Genode code could live in the same repo but I realized that I'm ripping out too much stuff to make it work.

It shouldn't be too long before I have something that sort of works, then I imagine there will be a lot of testing.

ehmry, 2015-03-28:
#### Status Update
I'm writing a server from scratch that will manage build processes and control all writing to /nix/store. This takes care of locking disipline and is a massive TCB reduction. Processes that evaulate the Nix language will link against slightly patched upstream Nix libraries, and a new libnixstore will implement the old API by wrapping the Nichts_store service. A muxer will probably sit between clients and the server to handle a job queue, and possibly network requests.

Right now I'm working on some filesystem stuff to make testing easier.

nfeske, 2015-03-31:
Thank you for the update! To me, the plan sounds tangible. Also, it is very good that you got in touch with the Nix developers.

Btw, I learned that you'll join this year's Hack'n'Hike. That is so cool! Looking forward to meet you in person. :-)

ehmry, 2015-03-31:
Yea, it gives me a timeline to get stuff in a good state and an excuse to travel.

ehmry, 2015-06-09:
##### Update

My strategy now is to port the nix libraries to interface with three servers, builder, store_import, and store_export.

Builder will execute build jobs, each job reference counted using signal contexts passed from clients. Jobs will be accounted solely to the builder's quotas so that clients can't interrupt jobs that other clients are also waiting on. Builder is waiting on the store_export server.

Store_import is a file system shim that clients use to import files to the store. Store_import simply exports the File_system interface, so it can be used programatically and or by any naive utilities that perform file system operations. For each top-level node the client creates in a store_import session, the shim creates a tree of nodes that hash file data as it passes to the storage backend below. When the client wishes to finalise an import, it creates a symlink at the location of the new import node. During the blocking symlink call the server calculates the hash and moves the node to its final location. Reading the symlink, which only exists at the shim, reveals the final location of the import. This method works well enough that I did all my testing simply using Bash and cp before writing the Nix library support.
Store_import is working and supported by the Nix libraries, but needs some security fixes.

Store_export will be a file system shim that reads description files from the file system shared between the three servers and constructs VFS sessions. These will be used by the builder to create build environments that contain the proper dependencies for each job but could also be used to export specialized enviroments to users.
For instance, a workspace that contains a toolchain and text manipulation tools, or game scenarios that contain game data and a shared engine. I haven't started on store_export, I want to get the builder working better first.

The biggest huddle remains how much of the upstream Nix libraries to patch, replace or drop.

nfeske, 2015-06-09:
The import procedure is very cool! Much better then finalizing the import by closing the file-system session (which would not be easily possible with Noux). Thank you for the insightful update.

ehmry, 2015-06-22:
To give another update, I have the Nix-repl command line interface working with cli_monitor's line editor rather than readline. Builder is starting children and reporting exits back to clients, but lacks proper job queueing.

I do need a component that enforces a restricted view of the Nix store, but I've realised Noux's VFS provides what I need if I create derivation instructions that configure Noux with the Nix langauge rather than translate traditional unix arguments to Noux arguments in C++. I can do that without any changes to the Nix libraries if I treat the list of environmental variables exported to builders as a map of ROM filenames to store filesystem objects, and generate config ROMs using the Nix expression language. I will still be able to reuse package expressions from the unix Nixpkgs by creating a simple argument translation layer in the Nix language.

The good news is that I will soon be writting much of the port as Nix language functions, and will be testing and developing day-to-day on my Genode laptop.

nfeske, 2015-06-22:
Thanks for the update and the good news! :-) Here at Genode Labs, more and more developers consider switching. @cnuke already did. So we are moving faster to Genode than I originally anticipated.

I recently reworked the line editor of the CLI monitor to support the increasing dynamics of CLI monitor. The new version is the one on my Turmvilla branch. It has not yet moved to staging. I hope that you have based your work on the new version already? If not, I'm very sorry for the inconvenience.

Admittedly, because of my lack of Nix know-how, I am not perfectly able to comprehend everything you write in the second paragraph. But I get that you have identified a good path way to go.

ehmry, 2015-07-20:
I have a good amount of the port implemented in a feature repo https://github.com/ehmry/genode-nix

My current task is writing build expressions and tests, and shakings all the bugs out of Noux and the file system.

nfeske, 2015-07-22:
I just browsed the new repository, which I greatly enjoyed. I think it is sensible of you to maintain the Nix line of work within a dedicated feature repo.

For executing the ```nix-repl.run``` script, I first missed the information about the Genode branch I should use. With genodelabs/master, I got compilation errors (symlink-related). I figured that they may supposedly be addressed by your symlink patches that I merged into genodelabs/staging yesterday. But the state of staging is still not sufficient to build everything (i.e., the nixstore) successfully. For now, while everything is still in flux, it might be the best to cite a branch of yours as reference in the README file.

ehmry, 2015-08-28:
I'm closing this issue is it is now handled in an external repo.

closed 2015-08-28 by ehmry