genodelabs/genode #1402: pistachio: Does not boot on platforms with +4GB of RAM

ssumpf, 2015-02-06:
Bender relocates boot modules at to the highest possible free physical memory region. Since core identity maps the modules, it might end up accessing kernel reserved memroy (>=0xc0000000).

nfeske, 2015-02-09:
@ssumpf Could you please add a link to the modified bender source code here?

alex-ab, 2015-02-10:
@nfeske: please replace the original commit by the new one.

nfeske, 2015-02-10:
@alex-ab I actually had to cherry-pick your patch atop of the original commit. I have squashed both into one commit and pushed the result to staging.

alex-ab, 2015-02-10:
@nfeske: please add also to staging - humanreadable message if this overlap of regions happens next time

alex-ab, 2015-02-11:
@nfeske: please add the fixup commit to staging "fixup nova: detect overlapping module regions"

chelmuth, 2015-02-16:
@alex-ab I double-checked that the sha1sum of _tool/boot/bender_ is as in _origin/staging_ this morning after the adventurous binary merge conflict. For the log it was 8dc9c194036f8040b52318c2cb9267b9e05cc41d.

alex-ab, 2015-02-16:
Confirmed.

commit 5b11e6ee (2015-02-10): bender: restrict relocation below 2G physical

Fixes #1402
closed 2015-02-17 by ssumpf