genodelabs/genode #1407: Noux: initial working directory

ehmry, 2015-02-12:
Noux should be able to set its intial working directory from a its config before starting its child. Without this a shell would be needed to change the directory for a programs that cannot do this for itself.

nfeske, 2015-02-12:
The current working directory is actually handled via an environment variable. Please have a look at ```ports/src/lib/libc_noux/plugin.cc``` and look for ```NOUX_CWD```. So maybe we are lucky and the initial working directory can be already defined by setting this environment variable in the noux configuration (similarly to how we use to define the TERM variable)?

ehmry, 2015-02-12:
Yes, NOUX_CWD is working for me, thanks.

closed 2015-02-12 by ehmry