genodelabs/genode #1410: hw & odroid_xu: network tests fail

m-stein, 2015-02-13:


m-stein, 2015-02-13:
I've fixed all compile and config problems but the network tests (*lwip*/*lxip*) get stuck after
```
[init -> usb_drv] Services::Services(): No <hid> config node found - not starting the USB HID (Input) service
[init -> usb_drv] Services::Services(): No <storage> config node found - not starting the USB Storage (Block) service
[init -> usb_drv] Services::Services(): No <raw> config node found - not starting external USB service
[init -> usb_drv] Enabled EHCI (USB 2.0) support
[init -> usb_drv] Using configured mac: 02:00:00:00:01:01
```
I assume the platform driver or USB driver implementation to differ between Arndale and Odroid-XU and my simple approach to merge both into Exynos5 is bad.

m-stein, 2015-02-16:
@ssumpf: My [branch](https://github.com/m-stein/genode/tree/1410_hw_odroid_xu_network_tests_fail).

skalk, 2019-03-06:
I do not think we concentrate on this anymore, and therefore will close it.

closed 2019-03-06 by skalk