genodelabs/genode #1411: vbox: improve virtual time accuracy

cproc, 2015-02-16:
Currently, the virtual time is calculated by converting the difference of the current and last read host time stamp counter values to milliseconds, which then get converted to nanoseconds. This conversion can have noticeable rounding errors (for example 9.999ms -> 9ms -> 9000000ns) and the errors add up, because the absolute virtual time value gets incremented by these relative values.

Better accuracy can be achieved by converting the difference of the current time stamp counter and an initial time stamp counter value directly to the absolute virtual time value in nanoseconds.

chelmuth, 2015-02-18:
@cproc your patch has issues with uint128_t on 32-bit systems. Could you develop a solution that does not need this data type or fix the problem differently?

cproc, 2015-02-19:
Fixup commit 69e5264276b3116699476ec9893af689850b761a uses 'double' instead of 'uint128_t'.

@chelmuth: please add to staging

cproc, 2015-02-23:
Fixup commit 42abac607fc9bc85e779195cd1b2e0dd2d80e867 uses the 128-bit type and functions provided by VirtualBox instead of 'double'.

@nfeske, @chelmuth: please add to staging

commit 01c27071 (2015-02-17): vbox: improve virtual time accuracy

Fixes #1411
commit 679ae1dd (2015-02-27): vbox: improve virtual time accuracy

Fixes #1411
closed 2015-02-27 by cproc