genodelabs/genode #1412: core: sanitize location information in cpu_session_component::affinity() that it matches physical location

alex-ab, 2015-02-18:
Currrently it is possible to set via the cpu_session()->affinity(...) almost arbitary coordinates which will be out of range of available physical cpus. Question, should this sanitization be done in the affinity() method or should each platfrom take care (hw, foc, nova, etc) ?

alex-ab, 2020-01-14:
@nfeske: may you please review dd7daf3ad5be5911a299abcbb4aa93ee964540b0 and 2e08ebab8d6d848b31ab6701cdd9c30150291c0f

alex-ab, 2020-01-16:
I updated the commits according to @nfeske comments: dd7daf3ad5be5911a299abcbb4aa93ee964540b0, 89f2665b9e281afd131dd7c42f81358f52a4beba and 3b593c1d81be3d31bad0ff104f58880388ff6b41

alex-ab, 2020-01-16:
Sorry, had to clean up left over in the commits, just see issue_1412 branch.

nfeske, 2020-01-16:
Super! Thanks for extending the test program also. I merged your commits to staging now.

alex-ab, 2020-02-17:
commits entered master

alex-ab, 2020-12-14:
@chelmuth: dea09d4cc685591d86b0215354802d32f6826984 contains the discussed fix

closed 2021-01-25 by alex-ab