genodelabs/genode #1416: jitterentropy port update

ehmry, 2015-02-20:
The jitterentropy collection allocator was hanging for me under certain conditions on NOVA, and updating the library fixed it for me.

Upstream changelog shows that improvments have been made: http://chronox.de/jent.html

chelmuth, 2015-05-12:
According to the chronox site [version 1.2.0](http://chronox.de/jent/jitterentropy-1.2.0.tar.xz) of the library is out. Retrospective, I wonder what made us hesitate to merge the update to 1.1.0, @cnuke?

cnuke, 2015-05-12:
There is no particular reason IIRC, it probably just slipped our mind because it is not marked as fixed.

ehmry, 2015-05-12:
I can bump this to 1.2.0 when I get a chance.

ehmry, 2015-05-12:
That should do it.

chelmuth, 2015-05-13:
Merged into staging.

chelmuth, 2015-05-27:
Made it into 15.05

closed 2015-05-27 by chelmuth