genodelabs/genode #1418: foc: thread test succeeds although not all threads got constructed correctly

alex-ab, 2015-02-24:
```
[2015-02-23 23:59:24] thread alloc failed
```

chelmuth, 2015-02-24:
Which phase of the thread test triggers the failed allocation? `test_context_alloc` creates many threads just to stress the context allocator and does not represent a real use case. IMO 64 threads is already a rather big number for one component.

alex-ab, 2015-02-24:
So, you mean we should just limit the test case up to 64 parallel starting threads ? Currently it could be 90.

chelmuth, 2015-02-24:
No objections from my side but I'm curious what @nfeske thinks about it.

nfeske, 2015-02-25:
I am fine with reducing the maximum number of threads in the test,

But I have another thought: If the test is indeed meant to stress the context allocator, wouldn't it be a good idea to test the corner case where no additional context can be allocated? This corner case could be provoked without the need to create too many physical threads. Instead, we can trigger it via the ```alloc_secondary_stack``` function. That said, I do not propose to extend the test right away. I just wanted to record the idea somewhere.

commit 71ee9ecb (2015-03-06): foc: increase maximal supported threads

Fixes #1418
commit d303a254 (2015-03-13): foc: increase maximal supported threads

Fixes #1418
closed 2015-03-13 by alex-ab