genodelabs/genode #1421: Improve 'Timed_semaphore' accuracy

cproc, 2015-02-24:
Currently, the timeout thread of the 'Timed_semaphore' increases a local jiffies counter by 10ms on each timer signal, which can be inaccurate if signals arrive later. Better accuracy can be achieved by calling 'Timer::Session::elapsed_ms()' instead.

chelmuth, 2015-02-24:
Doesn't your approach add significant runtime overhead as each component using the timed semaphore does not only unblock in 10 ms interval but also calls the RPC 100 times per second?

nfeske, 2015-02-25:
@chelmuth I am not too concerned about it, at least for now:

* The timed semaphore is used by only a few (large) applications like Qt5 or VirtualBox. If almost every component used the alarm library, I would share your sentiment but this is currently not the case. I think that the uses of the timed semaphore won't grow much in the future either. I.e., I am thinking of our transition towards the new server API. We will certainly never use a timed semaphore in those components.

* Given that the use cases of the timed semaphore are rare, one additional RPC every 10 milliseconds is not a huge load, especially when compared to the costs of delivering the timeout signal on most base platforms.

Still, I won't deny that there is room for improvement. The most obvious one would be to replace the periodic timeout by one-shot timeouts. The next timeout could be scheduled with the deadline of the next alarm. Of course, if a new alarm comes in, the programming of the next timeout must be updated (as the new alarm may have a nearer deadline than the existing ones). This approach has two benefits.

1. When using timeouts at a lower granularity than 10 milliseconds (let's say one second), we safe the costs for the polling.
2. When using the timeouts at a higher granularity than 10 milliseconds, the timeouts become more exact (presuming that the timer driver has a sufficiently fine granularity).

I propose to keep @cproc's patch and do my proposed one-shot optimization as a subsequent step. Should we open another issue for the additional step?

commit f04ef4f0 (2015-02-24): Improve 'Timed_semaphore' accuracy

Fixes #1421
closed 2015-02-27 by cproc