genodelabs/genode #1424: Xml_node does not like single -->

chelmuth, 2015-03-01:
I forgot to remove the closing part of a comment `-->` in my init config and xmllint did not whine. But, when starting the scenario I got
```
[init] C++ runtime: Genode::Xml_node::Nonexistent_sub_node
[init] void* abort(): abort called
```
For simplicity, I adapted the printf test config
```xml
<config>
	<parent-provides>
		<service name="LOG"/>
		<service name="RM"/>
	</parent-provides>
	<default-route>
		<any-service> <parent/> </any-service>
	</default-route>
	-->
	<start name="test-printf">
		<resource name="RAM" quantum="10M"/>
	</start>
</config>
```
Xmllint is right to stay silent as this is valid XML according to other online validators, e.g., http://validator.w3.org/. The `-->` needs to be interpreted as TEXT resp. CDATA. In our case this means the TEXT has to be ignored during parsing.

tomga, 2015-04-06:
This parser treats end of comment exactly the same way as it treats text in this place. Same exception is thrown when you put text in this place.

What is worse this parser does not throw exceptions consistently and seems to ignore tag content from any text to end of tag.

For following config there is no exception thrown and test works as expected:
```xml
<config>
    <parent-provides>
        <service name="LOG"/>
        <service name="RM"/>
    </parent-provides>
    <default-route>
        <any-service> <parent/> </any-service>
        sometext
    </default-route>
    <start name="test-printf">
        <resource name="RAM" quantum="10M"/>
    </start>
</config>
```
but if you move ``sometext`` before ``<any-service>`` exception is not thrown either but test fails with:
```
[init] test-printf: no route to service "RM"
[init] test-printf: no route to service "LOG"
```
which suggests that content after ``sometext`` is ignored silently.

Given that there is at least one open issue about xml parsing (#70) do you consider using external parser like suggested in this issue or maybe it is not acceptable due to some reasons like licencing or something else?

Another question is if there is some testing infrastructure for just testing code without all genode specific parts and running them natively on build machine? Given the complexity of debugging genode code running in qemu it would greatly simplify development of utility code like this xml parsing.

nfeske, 2015-04-07:
Let me clarify the role of the XML parser within Genode. The parser is merely there to provide low-level components with their configurations. It was created for this particular use case and does not even attempt to be generally applicable. For its design, the following considerations were taken into account:
* It must not be complex. Indeed, the parser comes in the form of a single header file (repos/os/include/util/xml_node.h) with less than 350 lines of code.
* It must not depend on a C runtime. Otherwise, each Genode-based system would depend on a C runtime.
* It should not dynamically allocate memory to make it usable in resource multiplexers where no anonymous memory allocations are allowed.
* It must be robust in the sense that it must not contain buffer overflows, infinite loops, memory corruptions, or any other defect that may make the program crash.

Given these considerations, other possible goals like expressive error messages, the support for non-existing use cases, or even the adherence to the standard are deliberately subordinated.

Your particular example mixes text and sub nodes within one node. Well, this is a case that never occurred in our configurations because it is no sensible use of XML. I wonder why anyone would want to do that in the first place. I think that any effort spent (and code complexity added) on the account of dealing with such use cases is ultimately wasted. As for the comment handling, one could argue that support for comments is not strictly required either, as comments could be stripped out in an offline step before booting the system. However, we still decided to support them to ease the interactive use of commented configuration data on the target machine.

Given this rationale, I fail to follow your proposal to replace the existing XML parser by another one. The only benefit would be the improved support of use cases that we don't have. But we would pay with an added complexity that is orders of magnitude higher than what we have now. This is unacceptable.

That said, let me reiterate that the XML parser is not meant as a general solution. It is specific for our domain. If you require advanced XML processing capabilities in one of your components, e.g., validating XML data against a dtd or schema, or mutating XML nodes, or using different character encodings, using a port of an existing XML parser would certainly be a good idea.

Regarding your question about debugging Genode code that is unrelated to device drivers or the kernel, I recommend you to run Genode on Linux during development and debugging. When a Genode system is started on the base-linux platform, each Genode component is actually a Linux process. So it is straight forward to use GDB (with the argument -p) for debugging.

tomga, 2015-04-07:
Thanks for detailed information about goals of this parser. My understanding about them came only from reading genode issues and in this one @chelmuth wrote:

> The --> needs to be interpreted as TEXT resp. CDATA. In our case this means the TEXT has to be ignored during parsing.

So I expected that text SHOULD be ignored and that current implementation tries to achieve this goal.

In #70 @abique - not me - proposed using external parser and there was no response for this comment so I couldn't know what do you think about it.

There is also #1230 in which even you, @nfeske, agreed to add complexity to parser to achieve more conformance with standard.

That's why it wasn't clear to me in what direction you plan to extend/fix this parser.

Now given that text in xml is not supported, do you think that exception about it should be consistently thrown? Currently it seems that it is thrown only from top level xml node. The patch probably would not be huge and I could possibly try to prepare one in my limited timee if you think it would be useful.

nfeske, 2015-04-08:
> Now given that text in xml is not supported, ...

That's actually not what I stated. Text in an XML node is supported (and used) but mixing text with sub nodes is not.

> Now given that text in xml is not supported, do you think that exception about it should be consistently thrown? Currently it seems that it is thrown only from top level xml node. The patch probably [...]

Since I fail to acknowledge the problem, I am not able see the value of a fix.

What I take from this discussion is that we should be more precise with documenting the role and feature set of the XML parser. I.e., the fact that the parser is not always able to identify sub nodes when they are mixed with text is not documented yet. Thanks for pointing me to this shortcoming.

I would still consider an investigation of the parser's behaviour when mixing sub nodes with text as worthwhile. Maybe there is still a bug lurking in the parser where it throws an exception where none is expected? E.g., does num_subnodes always returns a correct value, even in presence of additional text? If not, the subsequent attempt to iterate over the supposedly existing sub nodes may unexpectedly fail. More broadly speaking, a unit test for such corner cases would be valuable to have. I just dusted off an old test. Maybe it would be good as a starting point: https://github.com/nfeske/genode/commits/xml_node_test

tomga, 2015-04-08:
> That's actually not what I stated. Text in an XML node is supported (and used) but mixing text with sub nodes is not.
You're right. My mistake.

My conclusions are based solely from observing three different behaviours when inserting text in configuration above:

```
[init] C++ runtime: Genode::Xml_node::Nonexistent_sub_node
[init] void* abort(): abort called
```
when text is in place of ``-->`` from original example.

Success when inserted like in my example and
```
[init] test-printf: no route to service "RM"
[init] test-printf: no route to service "LOG"
```
when text is inserted like below:
```xml
    <default-route>
        sometext
        <any-service> <parent/> </any-service>
    </default-route>
```
which makes me think that ``any-service`` part is ignored in such case.

I'll extend your test with checks for such cases and expected content after parsing.

Anyway it is still not clear to me if you would like to ignore silently text in mixed nodes or throw exceptions in such cases. What would you expect it to behave? For me both solutions seem ok, although I typically prefer reporting errors. But then something sensible should be done with it in all places and I don't know Genode code well enough to perform a guess about it.

nfeske, 2015-04-08:
An exception should only be thrown on the attempt at access a sub node that does not exist. This may happen in two cases:
* Xml_node::sub_node is called with a node type argument and no sub node of the specified type exists. Normally, programs catch such exceptions to gracefully deal with situations where the XML data is incomplete.
* Xml_node::sub_node is called with an index that is higher or equal to the number of sub nodes present in the node. This case is normally not expected. E.g., when iterating over sub nodes, the usual procedure is to first request the number of sub nodes (via the num_sub_nodes method) and then iterate over the sub node indices up to the number of sub nodes - 1. Hence, the exception should not occur. Seeing init aborting while processing the XML data, however, makes me suspect that the num_sub_nodes method returns a non-zero number because it accounts for the sub nodes that are intermixed with the text. In the example, this would be 1. But when calling sub_node(0), the parser observes the text (which it does not expect) and does not even try to find a sub node following the text. It just throws an exception.

If my interpretation is correct (I have not looked at the code to confirm it), there are two possible fixes:
1. The num_sub_node function could always return 0 if it stumbles over text.
2. The sub_node(i) function skips text while searching for the node with the matching index.

Both fixes would be fine whereas I would intuitively prefer the latter.

tomga, 2015-04-13:
I've finally found a little time to experiment with this code. I haven't fixed throwing exception with mixed text and nodes yet but have another question.

To be able to attach to a process before it finishes I had to create some loop active loop in the code and I put a simple ``Xml_node("<config/>")`` in it. When I attached backtrace contained:

```
#0  0x000000000100b488 in Genode::Heap::_try_local_alloc (this=0x107cca0 <cxx_heap()::heap>, size=140, out_addr=0x400ffa50) at /data/projects/genode/genode/repos/base/src/base/heap/heap.cc:107
#1  0x000000000100b58c in Genode::Heap::_unsynchronized_alloc (this=0x107cca0 <cxx_heap()::heap>, size=140, out_addr=0x400ffa50) at /data/projects/genode/genode/repos/base/src/base/heap/heap.cc:147
#2  0x000000000100b6ae in Genode::Heap::alloc (this=0x107cca0 <cxx_heap()::heap>, size=140, out_addr=0x400ffa50) at /data/projects/genode/genode/repos/base/src/base/heap/heap.cc:197
#3  0x000000000102fb57 in malloc (size=132) at /data/projects/genode/genode/repos/base/src/base/cxx/malloc_free.cc:68
#4  0x00000000010303e0 in __cxa_allocate_exception ()
#5  0x000000000100035d in Genode::Xml_node::Attribute::Attribute (this=0x400ffb40, t=...) at /data/projects/genode/genode/repos/os/include/util/xml_node.h:91
#6  0x00000000010007d1 in Genode::Xml_node::Tag::Tag (this=0x400ffe28, start=...) at /data/projects/genode/genode/repos/os/include/util/xml_node.h:234
#7  0x0000000001001401 in Genode::Xml_node::Xml_node (this=0x400ffe10, addr=0x1032f4c <pseudo_end+1895> "<config/>", max_len=18446744073709551615) at /data/projects/genode/genode/repos/os/include/util/xml_node.h:498
#8  0x0000000001001be6 in print_xml_info (xml_string=0x1032e30 <pseudo_end+1611> "<config>  <program/>  sometext  <program/></config>") at /data/projects/genode/genode/repos/os/src/test/xml_node/test.cc:218
#9  0x0000000001001d12 in main () at /data/projects/genode/genode/repos/os/src/test/xml_node/test.cc:261
```

There is a call to ``__cxa_allocate_exception`` which contains call to ``malloc``. It seems to be invoked during every throw and it contradicts with need to have parser that does not dynamically allocate memory.

Is this behaviour (memory allocation during throwing exception) specific to ``lx`` target? If not how does it relate to requirement about not allocating memory during parsing xml? Should exceptions be used to control flow of execution in such cases?

nfeske, 2015-04-14:
This is indeed a tricky issue, which took us a while to deal with. In short, the C++ runtime normally expects a C runtime to be present. I.e., it calls malloc, string functions, and some I/O functions. This contradicts with our desire to omit the C runtime in the fundamental Genode components. For a more thorough discussion of the topic, please refer to Section 8.2 of the forthcoming documentation: http://genode.org/files/53bcb8e33fe6602fed25edc3c7b922c5/manual-2015-04-13.pdf

Fortunately, the allocation patterns of the C++ runtime are rather simple. Allocations (except for a few that happen at the initialization phase) are short-living and small. So it is possible to satisfy the dynamic allocation demands of the C++ runtime with a small dedicated heap partition. You can find the corresponding support code for the C++ runtime here https://github.com/genodelabs/genode/blob/master/repos/base/src/base/cxx/malloc_free.cc

Note that the malloc function you see in the backtrace is accessible only from the C++ runtime. It is not a public symbol. The C++ runtime and our support code are encapsulated in a single library (cxx). The malloc calls issued by the C++ runtime are local functions calls within this cxx library.

My comment above about our requirement to avoid dynamic memory allocations in resource multiplexers refers to the observation that typical libraries (including the C++ standard library but also utility libraries like XML parsers) do not consider dynamical memory allocations to be a problem. But in our case, this assumption does not hold true for some components. If you like to learn more about these considerations, I recommend you to take a look at the Section 3.3, and particularly 3.3.3 of the document linked above.

Could this explanation lift the clouds a bit? :-)

tomga, 2015-04-14:
I was expecting that this case was somehow handled, but finding it alone would take some time, so thank you for this information. It seems clear now. Design document looks very impressive. I'll read it later. Is there an html version available online? It would be more mobile friendly.

nfeske, 2015-04-14:
Although I plan to make the document available as HTML on the genode.org website by the end of May,  I'm not there yet.

tomga, 2016-02-14:
I wrote patches to this issue quite some time ago and I don't know if they are not commented because of licensing issue (fixed now) or because github doesn't emit informations about referencing issues. 

I'm just pinging it now, to get info about usability of those changes.

nfeske, 2016-02-15:
Thanks for the notification. Your patches have indeed gone unnoticed. GitHub does deliver an email for each comment placed at an issue, but not for commits that are popping up.

Thanks a lot for the fix and particularly for extending the test! However, the argument `eat_text_too` leaves me wondering because it apparently is not used in the code. To me it seems reasonable to unconditionally skip everything that is not a tag. So with your change, the method `eat_whitespaces_and_comments` could actually be renamed to `skip_non_tag_characters`, which includes whitespaces, comments, as well as arbitrary textual content (this intent could be stated in the comment of the method).

tomga, 2016-02-15:
I don't remember how what was my intention. Probably I introduced something, removed later and forgot to cleanup this part. I'll try to find some time to make changes you suggested in the next few days.

tomga, 2016-02-20:
I've made changes as suggested. Tests passed.

nfeske, 2016-03-10:
Thanks. I merged the result to staging. I squashed the first two commits into one, fixes two warnings in the test (caused by the use of %ld instead of %zd) and slightly adjusted the commit messages (writing "XML" upper case, removing the trailing dot). I hope you are fine with those minor changes.

tomga, 2016-03-10:
I'm fine with it. Glad to hear that it's useful.

chelmuth, 2016-05-31:
@nfeske please review my take on the XML-comment issue in 22eda64.

chelmuth, 2016-06-03:
Commit e3e2c99 is my next attempt to cleanly solve the issue. It adds two functions to `Token` for that purpose and safes `Xml_node::Comment` from the matching function. Opinions?

nfeske, 2016-06-03:
Beautiful! The new version nicely preserves the level of abstraction of the `Token`.

commit e5e4ae8d (2016-06-04): Parse XML comments independent of tokenizing

The whole XML comment has to be parsed as one XML tag to support strange
but valid combinations like

  
  

Fixes #1424
commit cf771c10 (2016-06-06): Improve tokenizing to support XML comments

The whole XML comment has to be parsed as one XML tag to support strange
but valid combinations like

  
  

Fixes #1424
closed 2016-06-22 by chelmuth