genodelabs/genode #1426: vbox: remove problematic assertion in recall handler

cproc, 2015-03-02:
The 'continue_hw_accelerated' assertion at the end of the recall handler can fail if the 'Timer' thread has set a 'forced action pending' flag in the meantime, which is not an error.

chelmuth, 2015-03-02:
I experienced this assertion too. Shouldn't we adapt the `Assert` condition according to your findings, i.e., assert that no flag beside _forced timer action pending_ is set?

cproc, 2015-03-02:
I'm not sure if it is really an error if any of the flags is set.

chelmuth, 2015-03-02:
So, we should find out ;-)

cproc, 2015-03-02:
Commit 5e43f80d073b1fa4e4c4b4cc581fc246ece0902a replaces the assertion with a debug message, which gets printed if any of the 'not yet verified as safe' flags is set.

commit dac3efcc (2015-03-19): vbox: replace too strict assertion with debug message

The 'continue_hw_accelerated' assertion at the end of the recall handler
can fail in situations which are not problematic, for example if the
'Timer' thread has set the 'VMCPU_FF_TIMER' flag in the meantime and
requested a recall afterwards. Since we don't know for sure if a recall is
requested for the other flags as well, the assertion gets replaced by a
debug message, which gets printed if any of the 'not yet verified as safe'
flags is set.

Fixes #1426
closed 2015-03-27 by cproc