genodelabs/genode #1429: part_blk: gpt support

cnuke, 2015-03-03:
At the last Hack'n Hike I added rudimentary GPT support to part_blk. The current implementation first checks for a GPT and then the MBR, if `use_gpt="yes"` is added to the configuration. Although that is convenient and this works fine when using well configured mediums, it might lead to problems when the medium contains a valid GPT as well as a valid
MBR but in this case the MBR should be preferred.

alex-ab, 2015-03-04:
:+1: Works well on one of my test machines having multiple OSes installed in various GPT partitions. I would like to see a option to enforce that it just use GPT, or just MBR.

chelmuth, 2015-03-27:
May we close this issue as it's labeled fixed and the commit above was merged to master?

alex-ab, 2015-03-27:
yes

closed 2015-03-27 by alex-ab