genodelabs/genode #1435: gdb: port preparation generates files into Genode source repository

skalk, 2015-03-05:
In general, generated files should not end up in Genode's source repository. In this specific case it is especially not wanted, if you share a contrib directory between several Genode source repositories. In case of our nightly buildbot: it re-uses the same contrib directory every night without the need to prepare again and again. The current state prevents several platforms from compiling `repos/ports/src/lib/gdbserver_platform`.

nfeske, 2015-03-06:
Wouldn't it be possible to perform the generating step at the build time? In my opinion, it would be preferable to avoid custom make rules in the port files.

cproc, 2015-03-09:
With commit 6fd56966f626063bcc820669269385fc419d4587, the files get generated at build time.

nfeske, 2015-03-09:
Thanks! I merged your new commit to staging.

skalk, 2015-03-11:
Using the new solution after preparing GDB I get the following error when compiling `run/gdb_monitor`:

```
  Library gdbserver_platform
    COMPILE  gdbserver_platform_helper.o
    COMPILE  linux-x86-low.o
    COMPILE  okl4_x86_low.o
    MERGE    libc_lock_pipe.lib.so
    MERGE    gdbserver_platform.lib.a
/usr/local/genode-gcc/bin/genode-x86-ar: i386-avx.o: No such file or directory
make[2]: *** [gdbserver_platform.lib.a] Error 1
make[1]: *** [gdbserver_platform.lib] Error 2
make[1]: *** Waiting for unfinished jobs....
    MERGE    libc_terminal.lib.so
```

cproc, 2015-03-11:
Commit beb001433df6d6f8eb85b65159a97f32a2763400 should fix the problem.

@nfeske, @chelmuth: please rename the commit message of the merged commit to 'gdb: save generated files in the build directory'

commit 39bee88e (2015-03-09): gdb: save generated files in the contrib directory

Fixes #1435
closed 2015-03-13 by cproc