genodelabs/genode #1437: vbox: wait longer for stable timer state

cproc, 2015-03-05:
Some functions in the time manager, for example 'TMTimerSet()' and 'TMTimerStop()' let VirtualBox abort with a failed assertion if the timer does not change to a 'stable' state after 1000 calls of a mixture of 'yield' and 'nanosleep'. On Genode, this happens sometimes when the 'EMT' thread is executing 'TMTimerSet()' and gets interrupted by the 'TAP' thread, which calls 'TMTimerStop()' and waits for the 'EMT' thread to finish setting the timer. Since the 'EMT' thread has the lowest priority, 1000 retries can be too few. Without the assertion, these functions would return an error code, which is often ignored by the caller, so it seems safer to keep retrying until the function can return successfully.

nfeske, 2015-03-06:
Wouldn't the excellent commit message a candidate for an in-source comment?

On another note, could you please rebase the patch on the current staging branch? Unfortunately, it conflicts with alex` recent patch.

cproc, 2015-03-09:
@nfeske: I've rebased the commit and added the comment to the patch file

commit 464a5586 (2015-03-09): vbox: wait longer for stable timer state

Some functions in the time manager, for example 'TMTimerSet()' and
'TMTimerStop()' let VirtualBox abort with a failed assertion if the timer
does not change to a 'stable' state after 1000 calls of a mixture of
'yield' and 'sleep'. On Genode, this happens sometimes when the 'EMT'
thread is executing 'TMTimerSet()' and gets interrupted by the 'TAP'
thread, which calls 'TMTimerStop()' and waits for the 'EMT' thread to
finish setting the timer. Since the 'EMT' thread has the lowest priority,
1000 retries can be too few. Without the assertion, these functio [... cut]
closed 2015-03-13 by cproc