genodelabs/genode #1439: wifi_drv: various fixes

cnuke, 2015-03-06:


cnuke, 2015-03-10:
@nfeske or @chelmuth please merge the latest wifi_drv fixes from https://github.com/cnuke/genode/commits/issue%231439_2 to staging. That should fix the problems observed on 32bit platforms.

nfeske, 2015-03-10:
I reviewed and merged the patch series into staging.

As a general guideline, I would suggest to remove commented-out code completely from the source code. I am referring to https://github.com/cnuke/genode/commit/d5d3b945461f96194db4e8442e838492115a04fe

chelmuth, 2015-03-11:
@cnuke again I'd like to express that I'm impressed you found the cause of wrongly-named 32-bit issue :+1: I'm still shattered GCC added this empty-struct extension as these are not valid C99. So, if anybody knows about a solution to successfully compile Linux kernel code _and_ force compiler errors on empty structs, I'd like to hear about it.

chelmuth, 2015-03-11:
Hm, should we enable `-Wc++-compat` in dde_linux?

commit f6b2fb9e (2015-03-06): wifi_drv: fix find_next_bit function

Among others, this function is used in the for_each_set_big() macro,
which is used when configuring the data rate tables. Therefore, this
fixes observed performance issues.

Fixes #1439.
commit 7923b287 (2015-03-13): wifi_drv: fix find_next_bit function

Among others, this function is used in the for_each_set_big() macro,
which is used when configuring the data rate tables. Therefore, this
fixes observed performance issues.

Fixes #1439.
closed 2015-03-13 by cnuke