genodelabs/genode #1443: hw: support for kernel-protected capabilities

skalk, 2015-03-10:


skalk, 2015-04-09:
@nfeske @m-stein : would you mind to review my current publicly available branch regarding this issue, and resp. put it to staging? Thereby, the state of the art of this issue does not detach from staging so much. Here is the current branch: https://github.com/skalk/genode/tree/issue%231443

nfeske, 2015-04-10:
Here are my comments:

* ```base: assign allocator and quota to platform pd```: The patch looks good but we will eventually also need a way upgrade the ram quota during the lifetime of the PD session. Of couse, this could be added as a follow-up patch later.
* ```core: remove global capability id counter```: Very good!
* ```base: remove local capability from generic base```: This change is very welcome and actually eases my life with my sel4-related work.
* ```hw: rename bin_* syscalls with delete_*```: I support the change but @m-stein should acknowledge it.
* ```hw: replace kernel's object id allocators```: The simplification looks good to me but @m-stein should review it.
* ```hw: directly reference kernel objects from core```: Just as we discussed before. No objection from my side.
* ```hw: construct kernel irq objects on demand```: I am fine with the change.

From my perspective, all the patches could be merged to staging as soon as @m-stein confirms.

m-stein, 2015-04-13:
Besides the minor inline comments, I'm fine with the commits.

skalk, 2015-05-20:
@m-stein : please review the last commit, which implements kernel capabilities

commit b96527f6 (2015-05-20): hw: kernel backed capabilities (Fix #1443)
commit d1870f3a (2015-05-21): fixup: "hw: kernel backed capabilities (Fix #1443)"
commit 1019cad5 (2015-05-26): hw: kernel backed capabilities (Fix #1443)
closed 2015-05-26 by skalk