genodelabs/genode #1444: vbox: support multi touch input events

alex-ab, 2015-03-10:
Enable support in virtualbox and our usb driver to process multi touch input events. That would enable us to gestures in (Windows) guests.

alex-ab, 2015-03-12:
This version based on staging works now reliable. Please review the input interface changes. Especially the signalling of the end of a touch event I don't like until now. Better suggestions are welcome.

alex-ab, 2015-03-13:
@nfeske, @chelmuth : changed the commits according to your comments

nfeske, 2015-03-13:
This looks very good. Unfortunately, commit c10a0df25f0ee1da9069aa1f875f27e2e110da86 does not apply on the current staging branch. Could you please rebase this commit?

alex-ab, 2015-03-13:
@nfeske: use f4a2bfaa722c2d6a982fb5837f796bc8f59e29f4

nfeske, 2015-03-13:
Thanks, merged to staging.

nfeske, 2015-03-16:
I get the following warning when compiling the window manager now:

```
In file included from repos/gems/src/server/wm/main.cc:32:0:
/repos/gems/src/server/wm/nitpicker.h: In member function ‘Input::Event Wm::Nitpicker::Session_component::_translate_event(Input::Event, Wm::Nitpicker::Point)’:
repos/gems/src/server/wm/nitpicker.h:471:11: warning: enumeration value ‘TOUCH’ not handled in switch [-Wswitch]
```

alex-ab, 2015-03-16:
@nfeske: use the fixup

alex-ab, 2015-03-16:
@nfeske: actually 7487c745ea7b823526fcd21c5007dedb4dcea060 is the better one

commit 53ee40a3 (2015-03-13): vbox: support multi touch input events

Fixes #1444
commit 2ad6a3b9 (2015-03-19): vbox: support multi touch input events

Fixes #1444
closed 2015-03-19 by alex-ab