genodelabs/genode #1446: nova: support sending genode signals via the kernel

alex-ab, 2015-03-12:
Currently Genode signal for base-nova are delivered via core. The current sending phase (omitting signal registration) protocol looks like this:

a) a signal source generates a Genode signal by sending synchronously to core a IPC
b) core notifies the receiver asynchronously via a semaphore up()
c) the receiver send an IPC to core to ask for the actual signal content (value and counter) and gets it delivered during the IPC reply phase

Beside all the book keeping in core, this requires at least 4 process context switches (a:1,b:1,c:2). Ideally this could be only 1 context switch, with a little bit help of your kernel.

alex-ab, 2015-03-13:
The current version works reasonable well and passes all tests so far. I think I will need to rework the base-nova specific part in Genode still a bit. Suggestions are welcome, please review.

The measurable improvements so far are that noux_tool_chain_auto speeds up by approximately 4.4 percent on 32bit and 2.5% on 64bit. The throughput test of signal.run (Test 3) is now able to send and receive about 32 million (32bit) and 37 million (64bit) signals per test interval - before the both commits above it was 0.94 million (32bit) and 1.3 million (64bit) on my test machine.

alex-ab, 2015-03-16:
@nfeske: commits e852fe3a3dc5ec0b430fa00f6f03efaa84f572ba and 22e312529513bf53712c729ee55e3769375be98a shows my refactoring attempts. please review. If you in general fine, I would squash both commits and move it actually before the nova commits and base the nova commits ontop of the squashed one

nfeske, 2015-03-17:
@alex-ab That looks very good!

alex-ab, 2015-03-17:
@nfeske: the patches got reworked and the link to the kernel branch is [r8_signals](https://github.com/alex-ab/NOVA/commits/r8_signals) and the actual commit is ["Extend semaphore with signal features"](https://github.com/alex-ab/NOVA/commit/196a945a2b5fd9343ac7c12b20fa7215bba592a6)

alex-ab, 2015-03-24:
The latest state can be found here [r8_signals_rcu](https://github.com/alex-ab/NOVA/commits/r8_signals_rcu). I had to adjust the kernel branch, since noux_tool_chain_auto on 64bit gets a OOM of the kernel. Turns out that the rcu lists get not processed quick enough, because of the too long grace periods (the idle CPUs dominate the overall grace period time since they slept 100ms). 

With the signal and rcu changes of the kernel and the fix of #1466 (avoiding some 64bit printf noise of 'madvise'), the tool chain numbers for our test machines improve from 66s/127s (32/64bit) runtime to 65s/76s.
If one disable one further printf noise in repos/base/src/core/rm_session_component.cc ("unmapping of managed dataspaces not yet supported") we get to 56s/66s - which is not to far off from the fastet kernel for this test - okl4 (32bit only) with 51s :-)

nfeske, 2015-03-24:
Very cool! I wonder, can you pinpoint the rather drastic change on 64bit to a particular change?

alex-ab, 2015-03-24:
The major improvement for 64bit is because of #1466. For 64bit the gcc issues tons of "madvice" calls causing a "not implemented printf". For 32bit there are no such madvice calls.

alex-ab, 2015-03-24:
Mmh, on the fuji it looks not sooo good, for the archives:

```
           fuji           macho
           old new        old new
okl4(32):  75s 75s        51s 51s
nova(32):  92s 85s        66s 56s
nova(64): 123s 99s       127s 66s

fuji : 2 cores - 4 hyperthreads - Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz
macho: 4 cores - 8 hyperthreads - Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
```

alex-ab, 2015-05-12:
@nfeske, @chelmuth : please review and consider 2b1634589871a2fb042a90fb12c3f491b7c8378f  for staging

chelmuth, 2015-05-13:
The issue is still in the works -> reopen.

alex-ab, 2015-05-13:
I tested it now with our relevant machines and it is working fine - so issue is fixed from my side. 56693e6bae79e2b116cb3b96989d6a224766dcaa

chelmuth, 2015-05-27:
The commits made it into 15.05.

commit 75c973b8 (2015-03-19): nova: leverage kernel to implement Genode signals

Fixes #1446
commit b50d707d (2015-03-27): nova: leverage kernel to implement Genode signals

Fixes #1446
closed 2015-05-27 by chelmuth