genodelabs/genode #1456: base: unification of irq_session interface - operate asynchronously

alex-ab, 2015-03-17:
Base-hw already has an own version of the irq_session interface operating asynchronously. Making it the general default behavior - being asynchronously - also for x86 kernels will ease supporting of MSIs (#1216) in the x86 platform driver (known today as acpi_drv and pci_drv) significantly.

nfeske, 2015-03-17:
As an intermediate step, I recommend to emulate the old (synchronous) wait-for-irq interface with the new interface - similar to how the timer-session's msleep function emulates the original (blocking) timer interface by using a client-local implementation (see ```Timer_session::Connection```). This way, you do not break compatibility during the transition. After the transition is complete and all Irq-session clients are updated to the new interface, the ```wait_for_irq function``` can safely be removed.

alex-ab, 2015-03-17:
Yes, in the first step I will resemble the base-hw interface - having still wait_for_irq and emulating there the old blocking behaviour in the client (instead of in core).

alex-ab, 2015-03-18:
With both commits the irq_session is working asynchronously for all kernels. The old wait_for_irq method is kept for the moment (to not touch the drivers for the moment.) Only for base-linux the irq_session_component.h is different, otherwise all use the some header in core. Now one could re-factor out the common parts of irq_session_component.cc from all base platforms and remove the necessity to use a proxy thread on foc_arm (was for the moment easier).

The other option is to stay more or less with this version and in the next step to move the shared irq handling out of core into the x86 platform driver. During this step the code gets nevertheless unified and in core only a simple (non-shared irq) version of irq_session_component will stay.

Opinions how to proceed ?

nfeske, 2015-03-19:
We should merge your commits to staging. But there is a point that needs further discussion/work:

I am not convinced that the ```Irq_session::Irq_signal``` definition belongs to the public API as it refers to (platform-specific?) implementation details of the signalling mechanism. From the documentation in the header, I cannot infer what this data structure and the accompanied RPC function (```signal()```) is good for.

alex-ab, 2015-03-19:
ask @m-stein - it is something specific to hw - not used by any other kernel

m-stein, 2015-03-19:
On hw, the ``signal`` RPC is used by an IRQ-session client to get the signal context that signals the IRQ. This is necessary by now as hw creates IRQ kernel-objects with statically bound signal contexts for all available IRQs on kernel initialization. However, this shall change in the future. IRQ kernel-objects shall be dynamically created and destroyed on the creation and destruction of IRQ sessions. That would also allow for the binding of individual signal contexts and hence, would make the ``signal`` RPC become unnecessary,

alex-ab, 2015-03-19:
But  is it really used already/currently in master ? I don't find a place where this information (stored in Irq_session_client.irq_signal is actually accessed, or ?

alex-ab, 2015-03-19:
Ok, I see it is used in the base-hw specific wait_for_irq implementation

cnuke, 2015-04-18:
@chelmuth or @nfeske please merge 8b97324 and 26020c7 to staging.

nfeske, 2015-04-18:
Done.

cnuke, 2015-04-20:
@chelmuth please merge b144b61 and 3ef91be to staging.

cnuke, 2015-04-21:
@nfeske could you please merge c1afc2e, 6a4e0a5, cfcd17d and 4a56bb5 to staging. The last commit should be moved before "os remove ATAPI driver" when the changes hit master.

commit e441cc75 (2015-04-17): base: make irq_session asynchronous

second step

options: factor out common parts of irq_session_component.cc
options: use on foc arm no proxy threads

Fixes #1456
commit ca68e60e (2015-04-20): base: make irq_session asynchronous

second step

options: factor out common parts of irq_session_component.cc
options: use on foc arm no proxy threads

Fixes #1456
commit faa25e1d (2015-04-23): base: make irq_session asynchronous

second step

options: factor out common parts of irq_session_component.cc
options: use on foc arm no proxy threads

Fixes #1456
closed 2015-04-23 by alex-ab