genodelabs/genode #1463: libc: reconfigure DNS resolver when network settings change

cnuke, 2015-03-20:
The libc DNS resolver only reads `/etc/resolv.conf`, which is generated automatically if DHCP is used, once when the application is started. If we change the network at runtime, it will still use the first configured nameserver.

chelmuth, 2019-03-14:
This was fixed by commit 74dcc7a3d57f91fd7382964ac1524ad622dd5f2d from issue #2861.

cnuke, 2019-03-14:
Well, AFAICT it still just reads the file only once - only the location of the file has changed.

chelmuth, 2019-03-14:
As this is implemented entirely in the contrib code, I wonder how long running programs in FreeBSD cope with such situations...

cnuke, 2019-03-14:
That is indeed the question but we still don't have an immediate use-case for the behaviour and probably don't want do fiddle around with libresolv anyway...

(I merely wanted to emphasize that #2861 is not strictly the fix for this issue.)

closed 2019-03-14 by cnuke

chelmuth, 2019-03-14:
For completeness: New versions of the FreeBSD libc implement time-driven configuration reload (see https://github.com/freebsd/freebsd/blob/master/lib/libc/resolv/res_state.c#L105).