genodelabs/genode #1464: hw: relative CPU-quota

m-stein, 2015-03-20:
CPU quota is currently donated to a thread on construction only by directly specifying a percentage of the quota of the according CPU session. In the future this shall be done by specifying any value on construction that is continuously set in relation to the values of the other threads at the CPU session. This requires dynamic re-configuration of the absolute quota of scheduling contexts in the kernel to adjust all threads at a session as soon the relative quota-value of one thread at the session changes.

m-stein, 2015-04-09:
@nfeske @skalk 

These are the main commits that solve the issue as it is (for early use (RPI) and review). However, still missing is:

* Every thread has a weighting > 0, also main threads (so all quota of a CPU session gets used as soon as its first thread gets created)
* RPC at CPU session that tells you how much physical quota the quota of the session is worth"

I'll add these the next days.

nfeske, 2015-04-10:
I like that the major portion of the change is the addition of the test! :-) Apart from the minor comments I annotated to the commits, I have no objections.

For the practical use (on the Rpi), won't I need the first of the two missing points to be covered? If I understand correctly, the change won't have any effect on the existing components (like the USB driver) otherwise.

skalk, 2015-04-10:
I have written few comments inline, beside from that I have no objections.

nfeske, 2015-04-10:
@m-stein Since you changed the Cpu_session interface, have you revisited all components that implement this interface? GDB monitor comes into mind. It would be worthwhile to grep the source tree at least for "Cpu_session_component".

m-stein, 2015-04-16:
@skalk, @nfeske 

I've updated my 1464_hw_relative_cpu_quota branch according to the suggestions and also added the feature of allowing only weightings > 0 and the Cpu_session::quota() call. The signal test succeeds on all base platforms with the patches and the CPU quota test now tests also the session quota. By now, I had no time to test more complex scenarios (GDB, noux) and were only able to re-base to master. I'll do that tomorrow.

m-stein, 2015-04-20:
@skalk, @nfeske

I've updated my 1464_hw_relative_cpu_quota branch. The commits are now tested with 'signal' on each platform, 'cpu_quota' on each HW platform except x86, 'gdb_monitor', 'noux_bash', and 'seoul'. On 'hw_x86_64', CPU quota isn't tested because the PIT with its low timeout limit distorts the results. But I assume that the PIT will be replaced as timing source in the near future anyway.

nfeske, 2015-04-21:
@m-stein Sometimes you use the term "weighting", sometimes you use the term "weight". Would you be ok, to consistently use the term "weight" in all places?

m-stein, 2015-04-21:
@nfeske No problem.

nfeske, 2015-04-21:
* "CPU session: apply quota via relative weightings": Apart from the minor style nitpicks that I annotated to the patch, I like it. I am not perfectly sure what to think of the use of the 128bit integer type for the computation. I would have tried to avoid it because I remember the use of such wide integer types on x86_32 as problematic. On the other hand, since base-hw does not support x86_32, this may be a non-issue.
* "hw: syscall for changing a threads physical quota": Looks fine to me.
* "hw: re-enable kernel tests with the new run env": Thanks for considering my suggestions.

m-stein, 2015-04-21:
@nfeske 
I don't like it either but the only way to circumvent this is to downscale the quota to an intermediate space (e.g. ms) and upscale it to tics afterwards. However, this would bring in additional rounding errors which became critical when testing small quota values. I explicitly moved it into an x86_64 dependent file to avoid compiler problems.

nfeske, 2015-04-21:
@m-stein Thank you for the rationale.

m-stein, 2015-04-21:
Updated the branch accordingly.

chelmuth, 2015-04-23:
@m-stein I pushed your commits to staging just now.

m-stein, 2015-04-24:
Looks good as far as I can see.

m-stein, 2015-04-24:
@chelmuth 
I've added a fix-up that solves the compile error in ``child.h`` when building QT5. I've solved it by further specifying the desired printf version in child.h. This doesn't of course solve the real problem (that someone seems to mess up the name spaces in a file that includes child.h) but if its not that important to you I'd like to refrain from searching for this place.

chelmuth, 2015-04-24:
The fixup looks completely okay.

chelmuth, 2015-05-06:
Issue is marked `fixed` and commits are on master.

closed 2015-05-06 by chelmuth