genodelabs/genode #1483: sd_card_drv for omap4 does not receive any interrupts

cnuke, 2015-04-16:
The driver waits endlessly in `_wait_for_transfer_complete_irq()` e.g.:

```
Genode 15.02-216-g84a98e1 <local changes>
int main(): --- create local services ---
int main(): --- start init ---
int main(): transferred 1021 MB to init
int main(): --- init created, waiting for exit condition ---
[init] Could not open ROM session for module "ld.lib.so"
[init -> sd_card_drv] --- OMAP4 SD card driver ---
[init -> sd_card_drv] CID: 0x02544d53 0x41303447 0x10179aeb 0x9700bc57
[init -> sd_card_drv] RCA: 0x1234
[init -> sd_card_drv] SD card detected
[init -> sd_card_drv] capacity: 3724 MiB
[init -> test-blk-cli] block device with block size 512 sector count 7626752
[init -> sd_card_drv] CID: 0x02544d53 0x41303447 0x10179aeb 0x9700bc57
[init -> sd_card_drv] RCA: 0x1234
[init -> sd_card_drv] SD card detected
[init -> sd_card_drv] capacity: 3724 MiB
[init -> test-blk-cli] reading block 0 - 7626751, 246 per request
[init -> sd_card_drv] bool Omap4_hsmmc_controller::_wait_for_transfer_complete_irq(): before wait_for_irq
```

cnuke, 2015-04-17:
Apparently the issue only exists on base-hw.

m-stein, 2015-04-17:
I'll have a look at it soon. Any deadline?

nfeske, 2017-06-29:
Works on the Pandaboard, covered by our automated tests.

closed 2017-06-29 by nfeske