genodelabs/genode #1486: pci: enforce policy on discovery of devices

alex-ab, 2015-04-21:


alex-ab, 2015-04-21:
The first version supports specifying the concrete bdf of a device or a device class, see the run scripts as example. The class names are unhandy for a human, should we introduce aliases for it ? And if yes, which one and where to add the aliases. (The pci_drv readme gets updated as soon as this becomes clear.)

nfeske, 2015-04-21:
The names "PNP0303" and "PNP0F13" are rather cryptic. Wouldn't a plain and self-explanatory "PS2" suffice?

alex-ab, 2015-04-21:
@chelmuth has another opinion about that

nfeske, 2015-04-21:
Instead of embedding a custom syntax in the bdf attribute, how about specifying each part of the device address as an individual attribute? Something like:
```
<pci bus="1" device="1" function="2"/>
```
This would alleviate the need to generate and parse a separate syntax in addition to the XML parser. The same syntax could be used to intuitively assign an entire PCI bus to a client:
```
<pci bus="2"/>
```

nfeske, 2015-04-21:
I will ask @chelmuth for his opinion then. ;-)

@alex-ab Regarding your question about named aliases, I would introduce them as soon as we see repeated hexadecimal numbers popping up in our run scripts. When I suggested the aliases, I was primarily thinking of the VESA driver that requires access to PCI devices of the class VGA. I guess that the assignment of USB controllers to the USB driver will be another recurring pattern. But I think it is fine to defer the introduction of the aliases until we feel the inconvenience.

chelmuth, 2015-04-21:
The PNP* strings are from the ACPI spec and indeed I proposed them to name the resources in a common way.

nfeske, 2015-04-22:
But I wonder, what is the benefit? I presume, the benefit should be that we do not need to invent names because the ACPI spec authors already did this for us. On the other hand, those cryptic names are hardly any better than magic hex values, don't you agree? In either case, one needs to consult the spec to interpret them.

Please consider the following points:

* We pick the names for us and the users of Genode. So why shouldn't we pick names that are already meaningful in our day to day conversations? E.g., HPET, RTC, PS2MOUSE, PS2KEYBOARD are precise enough for what we want to express and are intuitively clear.

* When looking at a configuration, e.g., when giving an example in the documentation or during a talk, which kind of name would you enjoy to present? A name that everyone in the audience readily understands or a name that requires further justification (referring to a spec that happens to be almost irrelevant to most people). I am actually a good example of a member of such a technical audience. Even as an OS developer for so many years, I have not looked at the ACPI spec even once and wondered what those strange names were.

* We are speaking about just a hand full of legacy devices. The situation is quite different from a case where we needed hundreds if name definitions. E.g., in the case of key code names, I supported the use of an existing definition because I could see the value. Here, the added value of sticking to the ACPI spec seems rather small. Or maybe I am just too shortsighted to see it.

Sorry for the wall of text. ;-)

chelmuth, 2015-04-22:
I'm convinced. Now, please tear down the wall ;-)
-- 
Christian Helmuth
Genode Labs

http://www.genode-labs.com/ · http://genode.org/ · /ˈdʒiː.nəʊd/

Genode Labs GmbH · Amtsgericht Dresden · HRB 28424 · Sitz Dresden
Geschäftsführer: Dr.-Ing. Norman Feske, Christian Helmuth

alex-ab, 2015-04-22:
ok - done, please review

commit 1f368f28 (2015-04-27): pci: enforce policy on device discovery

Fixes #1486
commit fcbee9fc (2015-04-28): pci: enforce policy on device discovery

Fixes #1486
commit 2fe0955b (2015-04-30): pci: enforce policy on device discovery

Fixes #1486
commit 7770a0fb (2015-05-06): pci: enforce policy on device discovery

Fixes #1486
closed 2015-05-06 by alex-ab