genodelabs/genode #1488: Deduplicate File_system servers

ehmry, 2015-04-23:


ehmry, 2015-04-23:
I have a commit for #1120 on that branch.

ehmry, 2015-04-24:
All the above servers pass libc VFS tests. If I tack 9pclient on the end of this its still a net reduction in code : )

nfeske, 2015-04-24:
Very cool! Thank you for taking the time to do this. I like that you paid attention to the include guards.

It seems that you missed to remove a few bits from ```os/src/server/ram_fs/```. I.e., ```util.h``` is still there. Removing it would further improve the stats. ;-)

Would you mind squashing the patch series into a single commit?

ehmry, 2015-04-24:
Done, ffat_fs and tar_fs still have their own util.h, but they're server specific.

ehmry, 2015-04-24:
I still need to write another server, I will try and minimize the packet handling when I do.

nfeske, 2015-04-24:
Great! I added your patch to staging. I hope you are ok that I slightly shortened the commit message to make it fit in the usual 50-character limit?

ehmry, 2015-04-24:
Thats fine, I thought git would complain if it got too long.

chelmuth, 2015-04-28:
@ehmry our buildbot choked on your patch with the following errors
```
> make -C build/nova_x86_64 server/fuse_fs
...
  Program server/fuse_fs/ext2/ext2_fuse_fs
    COMPILE  fuse_fs_main.o
cc1plus: warning: command line option ‘-Wno-implicit-function-declaration’ is valid for C/ObjC but not for C++ [enabled by default]
In file included from /plain/krishna/src/genode/genode_staging.git/repos/libports/src/server/fuse_fs/ext2/../fuse_fs_main.cc:29:0:
/plain/krishna/src/genode/genode_staging.git/repos/libports/src/server/fuse_fs/ext2/../directory.h:26:18: fatal error: util.h: No such file or directory
...
> make -C build/nova_x86_64 server/trace_fs
...
  Program server/trace_fs/trace_fs
    COMPILE  main.o
/plain/krishna/src/genode/genode_staging.git/repos/os/src/server/trace_fs/main.cc: In member function ‘virtual File_system::File_handle File_system::Session_component::file(File_system::Dir_handle, const Name&, File_system::Mode, bool)’:
/plain/krishna/src/genode/genode_staging.git/repos/os/src/server/trace_fs/main.cc:796:37: error: ‘valid_filename’ was not declared in this scope
```
Please provide a fixup commit on top of add6946c40030d54005691da8b15090e26654397. I would have done it myself but get lost among `valid_name()` and `valid_filename()` ;-)

ehmry, 2015-04-28:
@chelmuth done

chelmuth, 2015-04-28:
Did a short test build and added your fix to staging.

commit add6946c (2015-04-24): Move generic fs helpers to os/include/file_system

Fixes #1488
commit 55c0a947 (2015-05-06): Move generic fs helpers to os/include/file_system

Fixes #1488
closed 2015-05-06 by chelmuth