genodelabs/genode #1491: thread: prevent overlapping of stack top and first member in context

chelmuth, 2015-04-27:


chelmuth, 2015-04-27:
I admit that the proposed solution is just a kind of safety net if the stack top is dereferenced directly (like in thread.cc). On the other hand, the definition of *stack top* is somewhat vague: Does it point at the first word of the stack or just above the first word (with pre-auto-decrement in mind)? Therefore, I suggest to adopt my approach and have the first word of the stack as member in *Context* followed by other members, e.g. *stack_base* which was overwritten in *_alloc_context()* due to the overlap.

Additionally, I (again) notices that *base-foc/.../thread.cc* and *base/.../thread.cc* are almost identical beside the implementation of *Thread_base::myself()* and the stack initialization. But this is another issue...

alex-ab, 2015-04-27:
Is the reason that it only happened on the arm version because of the following line in thread.h:
```
addr_t stack_top() const                                            
115 .   .   .   {                                                                   
116 .   .   .   .   return ((addr_t)_stack & ~0xf) - Abi::stack_adjustment();       
117 .   .   .   }    
```

where Abi::stack_adjustment() is on x86/32 4 byte and x86/64 8 byte and on arm it is 0 byte ?

chelmuth, 2015-04-28:
You're right `Abi::stack_adjustment()` is involved in the issue but the root cause is that `_alloc_context()` does not adhere to pre-auto-decrement and just dereferences `stack_top()` as is. The stack adjustment was introduced to comply with the architecture-dependent stack-alignment constraints in the various ABI specifications (calling conventions etc.).

commit 5cffb671 (2015-04-27): Prevent overlap of stack and thread-context members

Now, Thread::Context provides the first word of the stack to prevent the
overlapping of stack top and the 'stack_base' member.

Fixes #1491
commit ce055045 (2015-04-30): Prevent overlap of stack and thread-context members

Now, Thread::Context provides the first word of the stack to prevent the
overlapping of stack top and the 'stack_base' member.

Fixes #1491
commit f20b6bdc (2015-05-06): Prevent overlap of stack and thread-context members

Now, Thread::Context provides the first word of the stack to prevent the
overlapping of stack top and the 'stack_base' member.

Fixes #1491
closed 2015-05-06 by chelmuth