genodelabs/genode #1492: hw: separate declaration from implmentation in kernel

skalk, 2015-04-28:
A lot of kernel related headers contain most of the function's implmentation. Thereby a lot of object member functions are implicitely inlined and increase the kernel/core's text segment. Moreover, every change of a function implies a lot of compilation units have to be recompiled. Last but not least it is often impossible to use certain functionality because of circular dependencies.

m-stein, 2015-04-28:
Good direction. I did some minor changes. Do you agree?

m-stein, 2015-04-28:
@skalk 
Good direction. I did some minor changes. Do you agree?

skalk, 2015-04-29:
@m-stein sure

chelmuth, 2015-04-29:
Merged to staging.

m-stein, 2015-04-30:
@skalk, @chelmuth 
I've added two fixes for the recent HW compile errors on build-bot.

m-stein, 2015-05-04:
@chelmuth 
Another fixup https://github.com/m-stein/genode/commit/845bc536945e872d692a5e4ab674c32273b5a31a. Thank you for the hint.

commit 7aa71fd7 (2015-04-29): hw: separate function declaration/implementation

Move kernel object functions from its headers to compilation units,
thereby reducing the kernel's text section, cache, and TLB footprint.

Fix #1492
closed 2015-05-06 by skalk