genodelabs/genode #1496: hw: timer test sometimes hangs on panda

chelmuth, 2015-04-29:
Seen last night by our buildbot, the timer test hangs in very rare cases on hw_panda. I had to loop the test and experienced the issue on the 13th retry with the following script
```bash
for i in $(seq -w 1 30); do
	echo
	echo "TEST: $i"
	echo
	make -C build/hw_panda run/timer || exit 1;
done
```

m-stein, 2015-04-30:
There are several sporadic issues with hw_panda currently (#1147, #982, noux) that might come from a problem with caches / branch prediction we investigated some time on last year but haven't found yet.

chelmuth, 2015-04-30:
My intention was to just record this rather sporadic issue. Thanks for cross linking to the other issues.

skalk, 2015-06-18:
I have tried the same loop that chelmuth proposed and cannot reproduce the problem with current staging anymore.

closed 2015-06-18 by skalk

chelmuth, 2015-06-18:
We did not change cache/branch predictor related code in the meantime and @m-stein's assumption did not apply, right?

skalk, 2015-06-18:
We did not change such code that is right, but honestly I do not know which changes affected this issue.