genodelabs/genode #1497: hw_usb_armory: USB in tz_vmm linux

m-stein, 2015-04-30:


m-stein, 2015-06-19:
I've decided to come back to journaling via github while being in the middle of this issue. So, what follows is a little summary about what happened until today.

After having base support and a native SD-card driver for USB-Armory, we noticed that USB (and therefore also network) didn't work in the Linux guest. However, for a cool demo of Genode hypervising a rich Linux, USB and network are mandatory. While trying to enable USB with our BusyBox Initrd, I decided to switch to one of the root file-systems that the USB-Armory Wiki suggests before digging deeper into the USB topic. Unfortunately, these file systems are pretty big and it isn't feasible to communicate them as Initrd as done with BusyBox. At the other hand, we don't want to give Linux the control over the ESDHC as we'll need it to be trusted for secure boot. Thus we decided to implement a stub-block driver as done in our L4 Linux in the ports-foc repository. In contrast to the L4-Linux driver, the VM driver shall draw on an interface of secure-monitor calls (SMCs). The SMCs are caught in the VMM and resolved to a block session that goes over a part_blk instance to our native sd_card driver. The part_blk service ensures that the VM can access only one "non-secure" partition while all other partitions on the SD card can be safely used by Genode.   

After fixing several bugs that made the ESDHC driver get stuck in tz_vmm, I've implemented almost all SMCs for the stub driver during the last weeks. When initializing, the driver registers a shared-memory buffer at the VMM that is currently allocated via kmalloc . This is sufficient as long as the shared-memory communication is going from VMM to VM only. Furthermore, it reads the device parameters for its device structure and prepares the interrupt that is given in the Genode config of the VMM. I've chosen the TVE interrupt that isn't used otherwise by now. The VMM injects the interrupt via the TZIC-SWINT register as soon as the block session has completed requests. Then, the IRQ-handler of the stub driver calls the "collect" SMC that starts communication of request replies from VMM to VM.

Today I've finished the implementation of the "collect" SMC. In contrast to the L4 stub-driver where "collect" directly calls the Linux "end-request" function for every request, in the VM, "collect" must be called multiple times if there are multiple requests because it can't call "end-request" itself. So the call communicates a request reply via shared memory and returns "1", indicating that there may be further requests. The stub driver then handles the received reply and calls "collect" again, and so on until the SMC returns "0". This works just fine for the very first requests and after fixing the boot args and a problem with the access rights, Linux indeed tried to mount the stub-block device. Unfortunately, it fails to detect the file system correctly. Hence, next, I'll have a look at the block data that reaches Linux.

nfeske, 2015-06-21:
Thanks for the write-up. A very interesting read!

m-stein, 2015-06-22:
Dumped the block data at my working machine, at the Genode SD-driver and in the guest Linux - it's all the same. Checked whether the guest Linux has been compiled with Ext4 support - yes. Simplified the constellation by removing part_blk and providing the whole SD card through the stub driver. This caused Linux to read the partition table with a correct result but mounting fails as before. Looked into some Linux code paths to get a better picture of why it fails. By the way, replaced my previous dirty solution of the device-name getter by implementing the corresponding SMC that returns the names of the VMM config.

m-stein, 2015-06-25:
Tried to run L4 Linux AHCI-bench. Noticed that there are problems with our test setting and the run script. Rejected. Changed the device name to the one we're using in L4 Linux to avoid problems with hidden heuristics. Switched back to our busybox initrd to analyze the block device with the command line via ``mdev``, ``/proc/*``, ``fsck``, and ``dd``. Found, that the device integration is all fine. But although the block data indeed is the same for low offsets, it jumps back to offset 0 after 16K. This seems to be the problem.

m-stein, 2015-06-25:
Assumed that it is a caching issue. Switched from kmalloc to dma_alloc_coherent for the shared mem between VM and VMM. The switch was a little bit more complex than thought as we need the alloc to initialize our device but the alloc needs a completely initialized device as input. There seems to be no other clean way to receive uncached mem. Created a dummy device and filled some members off the top of my head until the device could be successfully registered. Now the dma alloc succeeds but mounting still fails. So, unfortunately its not a caching issue.

m-stein, 2015-06-29:
Fixed a bug in the 64bit offset communication between Linux and Genode. Now the data is all correct even for requests greater than 16K but mount still returns -EINVAL. Started looking into the ext2 internals behind sys_mount trying to find the origin of the error code.

m-stein, 2015-06-30:
There was a problem with a compile unit that did not get updated. After clean and re-make, ``mount`` got farther till the first write attempt. The write attempt triggered a page fault. As the basis of the stub driver was taken from L4 Linux where Genode back-end and Linux front-end are in the same PD, the driver tried to directly write to the content pointer of the Genode Block TX. After interposing the shared buffer, ``mount`` and successive ``ls`` on the mount point succeeded.

nfeske, 2015-06-30:
Yay!

chelmuth, 2015-06-30:
Great!

m-stein, 2015-06-30:
Mounting and working on files still works with part_blk interposed again. However, when switching back to the Ubuntu rootfs from MMC, Linux now complains as follows:
```
EXT4-fs (sda1): mounting ext2 file system using the ext4 subsystem
EXT4-fs (sda1): warning: mounting unchecked fs, running e2fsck is recommended
EXT4-fs (sda1): mounted filesystem without journal. Opts: (null)
VFS: Mounted root (ext2 filesystem) on device 254:0.
mount_block_root: do_mount_root error 0  ext2
```

m-stein, 2015-07-01:
The problem with the devtmpfs was caused by insufficiencies in my rootfs creation routine. Fixed. Also started a backup of the resulting rootfs to get rid of the time intensive process when the driver crashes the partition. Furthermore, created a second partition to separate the secure Genode stuff from the Linux rootfs. Now, the main problem is that the USB armory resets often (likely because of some loose contact) so Linux never finishes initialization. Next goal is booting via JTAG.

m-stein, 2015-08-13:
As the micro-SD slot at our Mk0 board meanwhile is pretty unstable, I decided to switch to one of our Mk1 boards. The solderless pin header for the breakout pads isn't available in online shops currently, so I started creating my own solution. Dismantling the cables of a USB-to-TTL and somehow connecting them directly to the pads wasn't that successful. Thus, I soldered some pin pairs off a broken board to solder them onto the USB armory. It looks a little bit funny but it works.

m-stein, 2015-08-19:
While preparing a new SD-card because my previous one broke, I have further automatized this process. After an indeterministic amount of SMCs (usually about 1500), the non-secure world gets stuck. It always occurs after a series of some consecutive MMC IRQ injections and two consecutive collect calls from the stub block driver.

m-stein, 2015-08-26:
I've implemented that the general purpose LED of the USB armory is switched on when we enter the VM and switched off when we leave the VM. Also configured Linux in a way that it doesn't break this behavior (or that of ESDHC/UART) by writing to the IOMUXC. The LED stays silent from the point on when UART output ends. Started to port the system to the IMX53-QSB to be able to use our JTAG debugger without soldering. Currently, tz_vmm gets stuck at the QSB as soon the VMM gets started.

m-stein, 2015-09-01:
Thanks to Stefan, we have found that there seems to be a problem with the secure and non-secure world concurrently accessing the same UART. Thus, we have implemented an SMC that prints a character via the LOG session of the secure VMM. Then, stub-drivers for the early-printk back-end as well as for the back-end of the console of the guest Linux were implemented. Now all serial output of the VM goes through the VMM and the UART can be protected. As a result, the guest stays alive longer but it still doesn't finish its initialization.

m-stein, 2015-09-08:
There was a bug in the serial stub driver that made the guest get stuck. After fixing it, Linux got farther, and stopped again with systemd complaining that it couldn't mount tmpfs because a file sysfs was missing. The problem is somewhere in my rootfs creation routine. I've partclone'd the rootfs of a fresh usbarmory and Linux gets much farther now. At some point our ESDHC gets stuck with:
```
[init -> sd_card_drv] Write 0x3d6f5000 0x1000
[init -> sd_card_drv] completion host signal timed out
[init -> sd_card_drv] Write 0x3d71c000 0x1000
[init -> sd_card_drv] wait till issuing a new command is allowed timed out
[init -> sd_card_drv] Write 0x3d728000 0x1000
[init -> sd_card_drv] wait till issuing a new command is allowed timed out
[init -> sd_card_drv] Write 0x3d730000 0x1000
[init -> sd_card_drv] wait till issuing a new command is allowed timed out
[init -> sd_card_drv] Write 0x3d794000 0x1000
```

m-stein, 2015-11-16:
Fixed by [1497_usb_armory_demo](https://github.com/m-stein/genode/tree/1497_usb_armory_demo) together with [pre-built binaries](https://github.com/m-stein/genode_binaries/tree/master/tz_vmm/usb_armory) and contributional sources [uboot](https://github.com/m-stein/uboot/tree/genode_hw_usb_armory) and [linux](https://github.com/m-stein/linux/tree/genode_hw_usb_armory_tz_vmm).

@skalk: Please review.

skalk, 2015-11-17:
@m-stein except for the 'inject_irq' VM session extension, which we talked about offline, I'm fine with the changes

m-stein, 2015-11-20:
Applied suggestions by @nfeske and @skalk.

skalk, 2015-11-23:
@chelmuth @m-stein confirmed by my side!

commit 21427ca6 (2015-11-23): TZ VMM & USB Armory: Demo

* Provide paravirtualized block API for accessing the second partition
  of the block device that is provided by the ESDHC driver.

* Provide paravirtualized serial API for sending log-output over Genodes
  serial port.

* Use the latest Linux suggested in the USB Armory Wiki [1] when on USB Armory
  while still using the older vendor Linux when on i.MX53 QSB. I.e.,
  provide a device tree through RAM and a rootfs through the paravirtualized
  block device when on USB Armory while providing ATAGs and Initrd when on
  i.MX53 QSB.

* Switch on the LED on the USB Armory [... cut]
commit 59d2a83f (2015-11-29): TZ VMM & USB Armory: Demo

* Provide paravirtualized block API for accessing the second partition
  of the block device that is provided by the ESDHC driver.

* Provide paravirtualized serial API for sending log-output over Genodes
  serial port.

* Use the latest Linux suggested in the USB Armory Wiki [1] when on USB Armory
  while still using the older vendor Linux when on i.MX53 QSB. I.e.,
  provide a device tree through RAM and a rootfs through the paravirtualized
  block device when on USB Armory while providing ATAGs and Initrd when on
  i.MX53 QSB.

* Switch on the LED on the USB Armory [... cut]
closed 2015-11-30 by m-stein